ADTPro/adtpro #1: Issue Mac High Sierra

largejpeg, 2017-11-04:
Have there been any issues reported using ADTPro with Mac High Sierra?

I'm using High Sierra with ADTPro 2.0.2 and Java Version 8 Update 151.  ADTPro is saying there are no serial ports.  I checked the serial configuration setting and the port number is blank and doesn't give me any options to choose one.  I'm using a USB to serial adapter with a Prolific chip.  The adapter is showing up in the hardware system report and it even shows up as a modem connection under the mac's network settings menu.

Any suggestions?

david-schmidt, 2017-11-04:
I did have one person that had the same problem - the solution for now was to run the `ADTPro.command` invocation rather than using the pre-packaged app.  Our `<sarcasm>`friends`</sarcasm>` at Apple seem to have mucked about with the package structure, or have some other permission issue with the packaged app, so it's not invoking rxtx correctly. The .command, however, takes no prisoners and "just works."

Depending on your security settings, you can just double-click `ADTPro.command` that is in the same folder as the app.  Apple may not trust you or me, so they may require you to jump through some hoops to run the command via double click; but if you bring it up in Terminal, you can bypass their nannying completely and run it directly.

largejpeg, 2017-11-04:
That worked great.

"Nannying" is an appropriate word for what Apple is doing these days.  Simple tasks like this become complicated with "tricks" and "workarounds" to circumvent security measures.

Thanks for your help!

closed 2017-11-04 by largejpeg

iKarith, 2017-11-05:
It wasn't an Apple thing, it was an Oracle thing.  *sigh*

The right way to do this now (according to Oracle) to bundle the whole JRE
with your application.

Some folks figured this was dumb and tried to continue using something
called jarbundler.  Except Apple then decided that if you're using
jarbundler that you actually want to use the legacy (and unsafe) JRE
version 6.  The result probably doesn't work if you built it on something
later.

The solution to that is replace the jarbundler stub application launcher
with a shell script.  If you want to see how I did it with AppleCommander,
have a look at this:

https://github.com/iKarith/AppleCommander/tree/mac-unistub

I haven't PR'd this because … new versions of SWT don't work with
AppleCommander, and old versions don't work with Oracle's current JRE.
This is not something I want to fix now.  :)  But the jarbundler task works
just fine.  :)

Joseph

jacexpo, 2017-12-20:
I can confirm that this same issue existed for me under Sierra - but with the Oracle Java version 8 U 151.
The same resolution (running adtpro.command) was the solution so that i could see my serial ports.

chaimpeck, 2018-05-29:
Running `ADTPro.command` worked for me! Thanks for the suggestion.

KrisKennaway, 2018-06-16:
This was necessary for me as well -- perhaps this should be documented in the installation instructions?