ADTPro/adtpro #3: Apple IIGS post bootstrapping HOST TIMEOUT

jacexpo, 2017-12-21:
Apple IIGS Rom 01 
 Null modem cable built per spec http://adtpro.com/connectionsserial.html#MiniDIN8
 bootstrapping process followed exactly per bare-metal instructions http://adtpro.com/bootstrap.html#Starting_from_bare_metal

 Successfully able to "SPEEDBOOT" - and it successfully load all 3 stages without any issue. Per other discussions i have read, if the first part of speedboot works - but the other two fail - that likely indicates a cable/serial issue. I have not had even 1 issue with that part.

 Once loads the adtpro program though - the IIGS can not further complete communication with the ADPRO server. Whenever requesting a directory listing - the ADTPRO server will report that it has received the request but a HOST TIMEOUT immediately appears. 
 Same thing when a change directory is issued. The ADTPRO server will report that change directory was requested but a HOST TIMEOUT immediately appears.

Any ideas?

david-schmidt, 2017-12-21:
Yes, definitely.  This is electrical in nature - specifically a problem with handshaking lines.  And it can be due to a number of problems.  Off the top of my head:

1. The physical connection in the DE-9 end of things can be marginal.  I've had situations that were corrected by deflecting the male pins just a bit so that they make more positive contact in their female receptacles.  
2. There could be a problem with the USB-RS232 driver; Prolific and its many crappy clones are the usual suspects in this arena.  FTDI is consistently good in both hardware and driver support.
3. There could be a problem with wiring in the null modem.  It's imminently possible to misinterpret my instructions there.

But a full, working Speediboot means things are going really well.  It's unusual to have that combination of successful bootstrap followed by other unsuccessful communications.  Some automatic decisions can change between bootstrap and normal operations if you've got multiple serial adapters in your machine - but if you only have the built-in "slot" 2 MiniDIN8, that should not be the issue.

jacexpo, 2017-12-21:
Tried it on another different physical apple Iigs. Same symptoms despite this one having accelerator cards and a scsi card and other fancy stuff in it. The first one is bare bones.

I have confirmed that in the control panel all the serial port settings are at default.

david-schmidt, 2017-12-21:
Great!  So now you've eliminated the GS as the source of the problem.  Now, we just have the null modem cable, the DE-9 connection, the USB adapter and its driver still remaining as suspects.

jacexpo, 2017-12-23:
You will love this. In all cases - the null modem cable between usb adapter and apple II (the minidin8 one or the din5 )is never moved - never touched nor adjusted 
 
I have two macs - and a Windows 10 PC. Two Apple IIGS and an Apple IIc
With the ch340 usb adapter, windows 10 to apple IIGS speediboot works but no directory afterwards
With the ch340 usb adapter, windows 10 to apple iIc no real success likely due to hardware issue on iic (see post in CSA2)
With the ch340 usb adapter , mac 1 and mac 2 - nothing at all, even basic serial commands not working - despite the drivers available at https://kig.re/2014/12/31/how-to-use-arduino-nano-mini-pro-with-CH340G-on-mac-osx-yosemite.html

Bought an FTDI usb adapter online.
 Windows 10 to apple IIGS speediboot and directory and everything work correctly
 Mac 1 to apple IIGS speediboot and directory and everything work correctly
 Mac 1 to Apple IIc - speediboot loads first part via machine language commands fine, but the speediboot loader never starts
 Windows 10 to appleiic - speediboot loads first part AND the speediboot loader works fine. IIc hangs when ADTpro should start
 Windows 10 to appleiic - load prodos is fine, and loading adtpro serial program works fine in legacy loader method - but hangs at the 800g 
 Windows 10 to either apple IIc or Apple IIgs- using speediboot and  vsdrive - loads successfully - however after that - every time entering a command at the basic prompt results in a syntax error

So +1 on ordering a FDTI serial adapter, but there is something wrong seemingly with the vsdrive and whatever happens afterwards.

jacexpo, 2017-12-23:
Further testing shows that there is no problem with vsdrive on either iigs (256kb or 4.25mb models) - it is likely the iic that has issues.

david-schmidt, 2017-12-28:
Is there any other help you're needing on this issue - or is it all due to the IIc hardware?

jacexpo, 2017-12-28:
No other help needed on this thread- we can document that in the case when the bootstrap works yet the directory does not work - the solution was changing out the USB serial adapter from a CH341 to a FDTI

iKarith, 2017-12-29:
I had some trouble with the IIc doing a bootstrap. I had to increase the pacing delay a little regardless of serial to USB interface. Possibly related?

Joseph
Sent via mobile

closed 2017-12-29 by david-schmidt