ADTPro/adtpro #4: Missing Raspberry RXTX libserial

jacexpo, 2017-12-27:
While reviewing the code for the raspberry, i noticed that the RXTX libserial.so is missing from the distribution package.

The startup script expects
# For Linux, use this:
if [ "$OS" = "Linux" ]; then
  if [ -f /usr/bin/raspi-config ]; then
    export RXTXLIB=lib/rxtx/rxtx-2.2pre2-local/arm

However the package does not contain the arm folder. See the tar extract below.

ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/README
ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/RXTXcomm.jar
ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/i686-pc-linux-gnu/librxtxParallel.so
ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/i686-pc-linux-gnu/librxtxSerial.so
ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/mac-10.5/librxtxSerial.jnilib
ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/win32/rxtxSerial.dll
ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/win64/rxtxSerial.dll
ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/x86_64-unknown-linux-gnu/librxtxSerial.so

david-schmidt, 2017-12-29:
Sorry about that - the Pi isn't supported in the main branch of ADTPro (yet), so this will be of interest over on the Pi side of the world (https://github.com/RasppleII/a2cloud).

closed 2017-12-29 by david-schmidt

iKarith, 2017-12-29:
Do feel free to check out A2CLOUD.  :)  My plans with that are actually to make it do less customness though, rather than more.  The biggest weirdness is that ADTPro is an exclusively GUI program that has to be run under X, so we start up a background "fake" X server intended for running automated tests and the like and start it up using that.  Of course we now have no control over ADTPro (no signals or anything), so we just ruthlessly murder (force quit/kill) ADTPro any time we need to change something and start it back up again.

The other problem is that A2CLOUD tries to do a zillion other things as well.  I installs web browsers and chat clients and emulators and archivers and remote serial consoles.  Ideally, the goal here is to break those pieces apart so if what you want is ADTPro, you get ADTPro.  (You'll also get AppleCommander and Java because those are dependencies.)

Longer term roadmap is to do that background process thing more intelligently and offer some patches.  I just have not had the opportunity to research how best to do that in any kind of a cross-platform way yet.