ADTPro/adtpro #5: rxtx for new OS_ARCH

jacexpo, 2017-12-27:
May I please request some guidance setting up a new OS_ARCH for ADTPRO?
The platform is an OrangePi Lite - an arm7l clone.
Prior to installing the library - i get an error that the arch_type is invalid in the java startup screen -  the application does load, but without serial port support

As the OSTYPE isn't recognized - i have downloaded and compiled the 2.1.7 rxtx library. 
I perform the regular ./configure and then make which completed successfully and I copied the following into place.

-rwxr-xr-x 1 pi pi 38204 Dec 27 17:55 librxtxSerial.so

I added the necessary line to the adtpro.sh file right after the "else" line when it fails to detect the OS_ARCH.

 export RXTXLIB=lib/rxtx/rxtx-2.1.7-bins-r2/Linux/armv7l-unknown-linux-gnu

adtpro starts without errors or warnings, however, attempts to use the serial port fail with the error

I tried adding the raspberry pi to specify the serial port, which doesn't help.
 ADTPRO_EXTRA_JAVA_PARMS="-Dgnu.io.rxtx.SerialPorts=/dev/ttyUSB0"

My serial port is indeed detected as /dev/ttyUSB0

[    7.286938] USB Serial support registered for generic
[    7.287026] usbserial: USB Serial Driver core
[    7.291599] USB Serial support registered for ch341-uart
[    7.294059] usb 2-1.1: ch341-uart converter now attached to ttyUSB0

Anything else need to be updated?

jacexpo, 2017-12-27:
I can also add that by using minicom - a terminal program for linux - and connecting to ttyUSB0, and with the IIGS accepting characters via IN#2 - i am able to type commands on the apple IIgs, letting me know that the hardware piece - serial cables and all that - are functional.

david-schmidt, 2017-12-28:
I'm pretty sure that's all you need - update the .sh to know where to pick up your library, then all should fall into place.  Are you seeing the serial ports enumerated in the port selection dialog in the ADTPro server?

jacexpo, 2017-12-28:
Thats what i did (editing the script - as you can see posted above)  - yet that doesn't function. There are no serial ports enumerated

jacexpo, 2017-12-29:
Good news!
I was using the stable build of RXTX - rxtx-2.1-7r2
But as it wasn't working as described above - i downloaded rxtx-2.2pre2 and compiled that up. I then edited the adtpro.sh to point to the new library and i am happy to report that it worked like a charm.
Dave- shall i include the changes to adtpro.sh and the library so you can include in the next release? Maybe via email?

david-schmidt, 2017-12-29:
Well, actually... the Raspberry wing of ADTPro users kind of lives over on the a2cloud side of the world (https://github.com/RasppleII/a2cloud).  You will want to get in touch with Ivan on that project.  Dave Schmenk (of Apple II Pi fame (https://github.com/dschmenk/apple2pi)) is also a good resource for compilation on Pi-like platforms.

closed 2017-12-29 by david-schmidt

iKarith, 2017-12-29:
Ivan's basically retired from A2CLOUD/A2SERVER at this point, and we've never supported the Orange Pi (which is not really even remotely related to the Raspberry Pi—not the same OS, not the same developers, nor really the same community.)

It uses an Ubuntu system, I have no problem supporting that in the future and in fact want to make what we have eventually work pretty much anywhere you want to run it.  But for now Orange Pi isn't something we support at all.  In part because I don't have one.  (Nor ODROIDs, Banana Pis, TinkerBoards, nor any of the half dozen other little cell-phone-CPU boards people have come up with in recent years.  Again, not opposed to that changing.)

jacexpo, 2017-12-30:
I have now also compiled the rxtx serial library and thus ADTPRO successfully works on another new OS_ARCH

aarch64-unknown-linux

The armv7l java process would consume 100% of cpu causing slowdowns in the speediboot transfers (and some drops) . No such problem with this H5 based CPU -  

I am happy to provide these two rxtx serial libraries to the project. I am not looking for support of orangepi - just want to give back to the community and maybe someone else who will have this combination.

david-schmidt, 2017-12-31:
I appreciate the effort - I'll have to do some thinking on how the RXTX library can be better supported.  I am not excited about being forking it (if it were in github... it's been a while) and maintaining it separately from the main line.

jacexpo, 2017-12-31:
these are not forks, just new ostypes compiled from the existing 2.2pre2 codebase. All that would be done is a few more lines in the adtpro.sh and two new directories under lib/rxtx

iKarith, 2018-01-02:
You might consider having a chat with Rob about how he rebuilt
AppleCommander to use Gradle.  It dealt with the SWT which has the same
kind of problems RXTX does.

Joseph

david-schmidt, 2018-01-02:
The thing is... RXTX is the project that manages the content (or should manage the content) in lib/rxtx.  I see there is a project https://github.com/rxtx/rxtx that may be the thing to connect to (via Gradle, as Joseph suggests).  This (or someplace like it - whatever is the "most official") is the place that updates to RXTX should go, and ADTPro should consume them.

jacexpo, 2018-01-04:
I am sorry i am not communicating correctly to what i am proposing. The only hard part about compiling RXTX librxtxserial.so is getting the environment created for it - downloading the necessary includes and then making it.  I did this for the two architectures listed above. 

What i am proposing is that currently, you are distributing ADTPRO with these files for the rxtx-2.2pre2.local 

-rw-r--r-- 0/0            2722 2009-02-05 00:16 ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/README
-rw-r--r-- 0/0           60672 2009-09-17 04:19 ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/RXTXcomm.jar
-rw-r--r-- 0/0           45271 2009-02-05 00:08 ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/i686-pc-linux-gnu/librxtxParallel.so
-rw-r--r-- 0/0          146489 2009-02-05 00:08 ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/i686-pc-linux-gnu/librxtxSerial.so
-rw-r--r-- 0/0          173432 2009-09-17 04:19 ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/mac-10.5/librxtxSerial.jnilib
-rw-r--r-- 0/0          122880 2009-02-05 00:12 ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/win32/rxtxSerial.dll
-rw-r--r-- 0/0          128000 2009-02-05 00:12 ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/win64/rxtxSerial.dll
-rw-r--r-- 0/0          174170 2009-02-05 00:09 ADTPro-2.0.2/lib/rxtx/rxtx-2.2pre2-local/x86_64-unknown-linux-gnu/librxtxSerial.so

I am proposing me sending pre-built librxtxSerial.so files for the armv7l-unknown-linux-gnu and aarch64-unknown-linux-gnu so that they can be included  in the distribution and therefore anyone who happens to have either of these two types of systems ADTpro will work out of the box (with the appropriate if/else changes to ADTpro.sh file too)

if there is a new distribution of ADTpro planned/coming out - then these could be added. If not - then no worries!