ADTPro/adtpro #6: Can't get ADTPro up and running

rsutc, 2018-01-04:
1. The ADTPro disk I purchased does not boot on the Apple //g. I just get a "check startup device" error. Other disks do boot normally.

2. I have Mac)S 10.12.6 If I run ADTPro fromth applications folder, and press serial, I get the error message "There are no serial orgs or the serial library RXTX is not installed or visible. At the ten website there ar no instructions for resolving this error on a Mac.

3. If on the other hand I run your program from another partition, it either behaves as above, or appears to run correctly, but does not actually send anything to the Apple //g I.E. the progress bar shows on the Mac, but nothing ha[pens at the //g end. Hmmm. Further checking says a previously downloaded 2.0.2 copy is the one that does nothing, and a freshly downloaded copy that gives the RXTX error. Note that if I dump a copy of the older dowloaded one that only appears to work into the user/applications folder, it gives the error.

4. Altogether this seems to be a tangled mess, and I cannot spend any more time on it. Too bad because I was planning to use some old Apple stuff in a class I teach here at my university.

Please advise
Rick Sutcliffe

david-schmidt, 2018-01-04:
Rick -

The first thing to do is to disregard the .app icon in OSX 10.12.6.  Apple has nannied you into oblivion.  You need to double-click on the `adtpro.command` script in order to start and have access to your serial devices.  Then, we can look at the serial devices the ADTPro server has access to.

I can help with next steps after that's working - I'm unsure if you're interested in moving forward based on your comments, but it's quite possible to untangle everything if you're willing.

rsutc, 2018-01-05:
OK I will have another try starting there on Mon-Tue when I am back at the university. The program did not complain that it didn't have access and seemed to be sending data, but nothing got there--no text scrolling across the Apple screen. Also, the ADTPro floppy was apparently no good. Just to check I ran the ADTPro command (but not with an Apple connected here at home, and the Terminal said:

/Volumes/RSFiles/ADTPro-2.0.2/ADTPro.command ; exit;
rsutc$ /Volumes/RSFiles/ADTPro-2.0.2/ADTPro.command ; exit;
ADTPro Server version 2.0.2

Rick

david-schmidt, 2018-01-05:
Yep, that's all expected.  You need to run the command on your connected setup.

It turns out that you purchased the Apple /// version of ADTPro floppy disk.  I am never sure if people really mean to buy that or they do by accident.  It sounds like you did it by accident.  Fortunately, I always put the Apple II version on the flip-side just in case.  So try flipping the disk when you try again.

rsutc, 2018-01-06:
It was accidental, but then I am getting old and feeble minded. I tell my students I never make mistreaks.
'Course I also tell them "Today is Wednesday, and this being computing science, everything I told you on Monday is obsolete."

rsutc, 2018-01-10:
OK, I realized that  with the rs232 to usb cable I needed a drive, as the File/Serial Configuration was only giving me the bluetooth ports. so from the uGreen site, aI found the appropriate driver and installed. This initially gave me two more serial ports to choose from
dev/tty.usb/serial
dev/tty.cu/serial
picking either and clicking serial freezes the app and does not connect.
So I am worse off, not better.

david-schmidt, 2018-01-11:
You definitely will need (the correct) driver in order to use the USB/Serial port.  I don't recommend the UGreen driver, but instead getting it from FTDI:
http://www.ftdichip.com/Drivers/VCP.htm
And if you're using 10.9 or above, you'll want 2.4.2:
http://www.ftdichip.com/Drivers/VCP/MacOSX/FTDIUSBSerialDriver_v2_4_2.dmg

rsutc, 2018-01-12:
OK I to that package and ran the installer. I restarted the Mac, even though it did not ask me toThen I started up a fresh copy of ADTPro, activated trace, and restarted it. The only ports offered under the serial interface item were the bluetooth ones. Here is what was printed on the terminal screen:
args.length = 0
                                                                               
Broadcast Message from root@Ricks-MacBook-Pro-ElC.local                        
        (no tty) at 15:47 PST...                                               
                                                                               
Communications lost with UPS                                                   
                                                                               
SerialTransport opened port named /dev/tty.Bluetooth-Incoming-Port at speed 115200.

rsutc, 2018-01-12:
Perhaps you could tell me what is supposed to be installed and where so I can check.

david-schmidt, 2018-01-12:
What OS/version do you have?

david-schmidt, 2018-01-12:
Did you install version 2.4.2 of that driver?  

Can you go to the system profiler (Apple menu/upper-left corner, About this Mac, More info) and see if FT232 shows up in your USB list?
[image]

rsutc, 2018-01-12:
Nope, it does not

david-schmidt, 2018-01-12:
Ok, then you don't have a driver installed that can communicate with the adapter.  As far as I can tell, you need 2.4.2 based on the level of OS you have.  Once the correct driver is installed, it'll show up in that list.

rsutc, 2018-01-12:
Hmmm. well, I did run the 2.4.4 installer, and when I look at Library/Extensions, I see
FTDIUSBSerialDriver.kext with a date of May 9,2017.
Should I delete this and run the installer again? Or is there also a rat file to be concerned about. What all is the pkg supposed to install?
the disk I installed from says:
FTDIUSBSerialDriver OS10.9_v2_4_2.dmg

rsutc, 2018-01-12:
That should be rct file. Spell correctors.

rsutc, 2018-01-12:
Here is the terminal listing of that file
drwxr-xr-x   3 root  wheel   102 May  9  2017 FTDIUSBSerialDriver.kext
the permissions and user/group are identical to everything else in that directory

david-schmidt, 2018-01-12:
[You can go back and edit comments... that helps.]

I'm not an expert on uninstalling Mac drivers, I'm sorry to say.  But cleaning out any unnecessary drivers that have to do with this (including earlier failed installs) would be a good idea, then starting fresh.

seanm, 2018-02-20:
@rsutc if you open Terminal and type 'kextstat', is "com.FTDI.driver.FTDIUSBSerialDriver" listed?  If not, the FTDI driver did not load.

david-schmidt, 2018-05-26:
Closing for now - feel free to re-open when/if necessary.

closed 2018-05-26 by david-schmidt