ADTPro/adtpro #7: USB Serial Port doesn't send anything during Bootstrapping?!

McNugget6750, 2018-01-06:
Hi there,

Thanks for this effort of making old Apple computers accessible again.
I found two Apple /// in the dumpster next to my apartment and I'm trying to get them to work.

I entered your machine code program and it reports HI. Unfortunately, neither my Windows 10 PC nor my Ubuntu laptop want to send anything using your software. It reports, the port was successfully opened but nothing is being transmitted. I checked with HTerm and I can send bytes to the Apple and I see the screen characters flickering around. But using ADTPro, nothing happens, not even on my scope or on the LEDs of the USB to Serial converter.

Any ideas what's wrong? Two systems running Java and none of them transfer any data?

Is there a way to transfer the ADTPro software manually using a different terminal program?
If so, where can I find the RAW file?

Thanks a lot,
Timo

McNugget6750, 2018-01-06:
Baby steps:

- When I use ADTPro and my serial adapter without the APPLE /// connected, should I see TX activity on the serial lines from the PC side?
- Do you ever request data from the APPLE ///? i.e. is the APPLE /// ever going to send data or is it just accepting data in the first stages of the program?

david-schmidt, 2018-01-08:
Hi, Timo -

That's an interesting set of circumstances there.  The symptoms point to the ADTPro server not really being able to open (or not being pointed at) the correct serial port.  How sure are you that the COM port it's using is the correct one (and is the same one that you use your terminal program with)?

To answer the later questions - I'm not sure if you'd see TX activity without a connection or not.  I don't have a breakout box myself, so I don't know what the electrical characteristics are offhand.  And during initial bootstrapping, the Apple /// doesn't send data back to the host until the first chunk has been received and is executing:
http://adtpro.com/protocol.html#SOS

There's not an easy way to send the raw files themselves - and even if you did, you'd still need the basic functions of the ADTPro server to do anything useful from there anyway.  So the thing to do is figure out why it's not really connecting to the serial port (or the serial port you think it's connecting to).

McNugget6750, 2018-01-12:
Hi David,

Thanks for getting back to me. Yes, I'm pretty sure the answer to all your questions is yes. Com5 in all terminals and sttyUSB0 on Ubuntu.
Anyways, I have one computer with a native serial port, the raspberry pi 3. Unfortunately, ADTPro doesn't start on the Raspberry Pi 1 due to memory limitations. I haven't tried the Pi3, yet. Maybe that solves it. 

One additional information: I don't have genuine FTDI chips. But I have good experience with my clones so far. Your RXTX library seems to be special and only wants to talk to genuine chips even though this is a slow speed application.

I'll let you know if I make any progress.

david-schmidt, 2018-01-12:
I can speak with more than 10 years of experience with every kind of USB adapter out there - nothing beats the real thing.  ADTPro puts a heavy strain on the serial subsystem with rapid data direction switches, and clone chipsets are the only ones that ever cause people trouble.

Just to clarify: I don't run/control/own RXTX; I simply use it for its universality.  And it does not have any predilection for one chipset over the other.  It just makes native calls to the operating system out to the driver.  If the driver or the chipset isn't reliable, RXTX won't be reliable either.  Again, genuine Prolific and FTDI chipsets have proven over time to be the leaders here.  I'm biased against Prolific because of the dubious things they've done in their driers to stamp out cloners, but I at least sympathize that cloners ultimately make them (and by extension) me look bad.

McNugget6750, 2018-01-12:
No worries, you don't look bad because some library can't work with cloned chips. However, I believe it's a library issue and not a chip issue. It's a hypothesis but since I worked with these chips for a while and was very happy (FTDI RS232RL clone and CH340) I doubt that they don't talk at all.
I believe RXTX might be trying to use a configuration the FRDI clone does not support which causes them to stop sending data.

An observation: When I close your software, the chip sends out a short burst of data. I didn't log that but I saw it on the scope. Maybe the buffer never gets flushed.

david-schmidt, 2018-06-01:
Doing some maintenance on issues - I don't think there's anything left to answer on this one, so closing.

closed 2018-06-01 by david-schmidt