ADTPro/adtpro #8: On macOS: Legacy Java SE 6 runtime is required, despite Java 8 already being installed

seanm, 2018-02-20:
When I launch ADTPro 2.0.2 on macOS 10.12.6, I get an error saying:

"To open "ADTPro-2.0.2.app" you need to install the legacy Java SE 6 runtime."

Presumably I could of course install the legacy Java 6 runtime, but I already have Java 8 installed.

The Java SE 6 runtime will presumably stop working with some future version of macOS, so it would be nice to support current Java eventually.

iKarith, 2018-02-21:
We also had this problem with AppleCommander.  Ultimately the entire build
system was rewritten using gradle rather than using ant.  I thought this
was going to make an already complicated mess thanks to SWT even more so,
but it actually really made everything much easier.

I'm not exactly an expert on gradle, unfortunately.  I can use say that
once the initial conversion was done, I had an easy time getting it working
on macOS and Linux.

Joseph

david-schmidt, 2018-02-21:
Apple has not been kind to you.  But you can get around their shenanigans by running the .command file rather than the .app.

iKarith, 2018-02-21:
I thought there was already a bug for this to be marked a duplicate, but it appears not.

david-schmidt, 2018-02-21:
There is - issue #1 .

iKarith, 2018-02-21:
Oh, that's my fault though.  I commented about the Java issue there on a bug about something else rather than reporting it separately.  So when the serial port issue got sorted, the issue got closed as resolved.

david-schmidt, 2018-06-01:
I'm looking at updating the app bundling process to cure the Java SE 6 issue (maybe bundling a JRE since folks can get all crazy and try to use Java v10) and get RXTX out of nanny jail with Apple.  Issue #26 is being used for that.

closed 2018-06-01 by david-schmidt