ADTPro/adtpro #9: Can't get my Apple IIc to communicate with Windows 10

MSoliven, 2018-02-25:
My setup:
- I'm using an original Apple IIc (peek output is 255)
- Connected the null model cable (purchased from Retrofloppy) to serial port 2 (Modem).
- RS-232 to USB port on my Windows 10 laptop

Method 1 (Using the client disk):

Server:
- When I ran the ADP pro's serial configuration, the ONLY port option that I can use is "COM3" defaulted to 115200 speed.
- The program exits as soon as I click the Serial button

Client:
- Picked (G) option to configured the COMMS DEVICE to "SSC SLOT 2". The other options are defaulted.
- Picked (R) to Receive, entered the filename, but I get a "HOST TIMEOUT" almost immediately.

Method 2 (Bootstrapping)

Server:
- Same as described in Method 1

Client:
- Rebooted, control-reset to get to the command prompt
- Typed "IN#2", to get the "?" promt
- Typed Control+A and then 1-4-B

I can't pick COM2 from the server host application. The server Java application ends as soon as I click the serial button. The Bootstrapping tab is never enabled.

Please help me troubleshoot this.

Thanks!
Menard

david-schmidt, 2018-02-25:
Hi, Menard - 

Did you download and install the USB driver for the RS232 adapter?

MSoliven, 2018-02-25:
Hello David,

Not on Windows 10. I switched to an XP machine and I saw that there was a problem with the driver.
I'm able to send data now. Can you confirm that I should be able to use this on a Windows 10 host?

Thanks,
Menard

david-schmidt, 2018-02-26:
Menard -

Yes, Windows 10 should not be the problem.  But for one reason or another, you Widows 10 machine does not have the proper driver installed for the adapter.  When it does, you'll see a new COM port show up for it.

iKarith, 2018-04-14:
If the driver for Windows XP works, but it doesn't for Windows 10, I suspect a clone chipset driver.  @MSoliven, did you resolve this issue?  If not, do you have a way to verify whether your USB-serial adapter is using a non-cloned chipset?  The makers of the authentic chips used in these USB-serial drivers have had a lot of trouble with cloned chips.  Prolific has made their drivers simply refuse to talk to fake chips.  FTDI actually at some point made their drivers brick devices using fake chips(!).

Unfortunately authentic FTDI-based devices are a lot more available than authentic Prolific, so I wind up having to recommend them despite really preferring not to.

Nicholas042893, 2018-05-10:
Unfortunately, I found my way here because I'm having this issue where ADTPro closes as soon as I select "Serial" for the connection option. 
I ran ADTPro on Windows 10 with the USB to serial Trendnet (Prolific chip) adapter that I've used the last couple years, and with the same drivers and it worked fine as recently as last year. I tried using it again with a fresh installation of Windows 10, Java RE 10, same drivers and the last release of ADTPro tonight and although COM3 is recognized as it should be, every time I click "Serial" it closes the program. I tried it on my Windows 10 laptop with a driver downloaded from Prolific rather than Trendnet this time, and COM3 shows up like always, and upon clicking "Serial" to connect it also closed the program. It doesn't seem to matter who I get the driver from or what version of the driver I'm using, I get the same results. 
On my MacBook Pro I was able to get ADTPro running and get a disk made using the same adapter but I'm running into other issues with that setup.

david-schmidt, 2018-05-10:
@Nicholas042893 - You're seeing what you're seeing because your Windows 10 machine is crashing when we try to talk to the serial driver.  One of several software-related things is wrong.  It could be that you've got a 32-bit version of Java running on a 64-bit OS (not super likely since it sees COM3 in the first place), or the driver is dodgy, or the chip in the adapter is dodgy as a potential clone of a legitimate Prolific chip.  I've long-since given up on Prolific since neither they nor their cloners can be trusted.  The arms race they wage has left no winners.

Nicholas042893, 2018-05-10:
Thanks for replying! Well, I'm using the same adapter with my MacBook Pro with drivers form the same place I obtained the Windows drivers today, and it works fine on my Mac (I just had to shorten some disk image names in the directory). Disk transfers and comm. are perfect. 

I've tried the driver I had when I had stable success with this same adapter on Windows 10 in the past and I've used the latest drivers downloaded from Prolific and Trendnet and all give the same results, seeing COM3 and crashing when clicking "Serial"

I might venture to guess it's to do with Java then, since Java is the only thing I haven't tried a different version of other than the latest RE 10 on Windows. I'm using the legacy RE 6 on my Mac and I honestly don't remember what version I was using in Windows at the time I had ADTPro working... I do have the 64 bit version running on 64 bit Win 10 though.  I'll work on it again tomorrow and see if Java version is the issue!

david-schmidt, 2018-05-10:
You can get some details of the OS crash by bringing up a command window and manually running `adtpro.bat` right from there.  You should get a screen dump of what happened to Java from there.

Nicholas042893, 2018-05-11:
@david-schmidt Okay, so I tried a direct execution from CMD and I attached a screenshot here, but it didn't actually log anything that I saw. So I went into the Java console and took a screenshot of the log shown in the second attached image. 
I uninstalled the latest version 10 and installed legacy version 6 of the Java SE R.E. and now when I click on "Serial" in ADTPro it works perfectly fine! It seems to be that at some point, a version of Java after 6 is causing a break in functionality. Hopefully that log can help determine what the issue is, but Java 6 on Mac OS 13 and Windows 10 is the current working set up I have here. 
[image]
[image]

david-schmidt, 2018-06-01:
Doing some maintenance on issues - I don't think there's anything left to answer on this one, so closing.

closed 2018-06-01 by david-schmidt