ADTPro/adtpro #10: ADTPro Not Seeing Disk ii in Slot 6

Dudesrug, 2018-03-04:
Hi All,

I just purchased a used Apple iie on ebay, along with a Disk ii for it.   The only 5.25 disk I have for it is a DOS 3.3 System Master, but it doesn't want to load when I boot the computer.  The drive spins and it's red light lights up, but there is no indication that it is loading anything from the disk.

I am able to get ADTPro on my Windows 7 laptop to talk to the Apple via serial cable, and have successfully run it and sent Speediboot to the Apple.  However, when I choose "F" to format a disk, it does not recognize that I have a drive at Slot 6 Controller 1.  It only lists Slot 3 and Controller 2, even though nothing is plugged into slot 2.

I have tried moving the Disk controller card to other slots but it still behaves the same.  I have also cleaned the Disk ii head.  

Anybody run into this?

Thanks,
Randy

cbmeeks, 2018-03-05:
Is it an original Disk 2 with the ribbon cable connection?  If so, is the ribbon cable properly connected to the disk controller card?

david-schmidt, 2018-03-05:
If ADTPro (and thus ProDOS) doesn't see the floppy controller, you might be in trouble.  But then if the controller spins the disk on power-up, that indicates it's at least partially alive.  This isn't an issue with the disk drive so much as it is with the controller itself.

One twist - there was a version of ADTPro prior to 2.0.2 that didn't "see" a drive unless it had good media in the drive; you don't happen to be feeding it something other than version 2.0.2, are you?

Dudesrug, 2018-03-05:
Hey, thank you cbmeeks and David for the prompt responses!

I do have an original Disk 2 with the ribbon cable.  I bought it separately from the Apple 2e and installed it myself, and I have tried reinstalling the controller card and re-seating the ribbon cable into the controller card several times.  I did notice that a couple of pins on Connector 1 on the controller card were bent an I had to bend them back into alignment before I could get the ribbon cable connected.  

I am using ADTPro 2.0.2, and the drive does spin with red light on and chugga-chugga-chug when I power up the computer and do PR#6.   So, as you observe David, the controller and drive are at least partially alive.

I also noted that once I've loaded ADTPro onto the Apple, even though (F)ormat and then (R)efresh does not reveal my drive on slot 6, if I go to the (Q)uit screen I can then hit tab and it will display Slot 6 Connector 1.  I can't find anyway back to the main menu from that point.

I have a second drive coming in the mail, which I had planned to purchase anyway, but I'm thinking I may need to pick up a replacement controller card.

Hey thanks again for the responses, and thank you for putting together the website and ADTPro!  I'm so happy that there is still a community out there dedicated to the Apple 2.  For me it was my first computer and I finally decided to pick one up and re-visit it.

I'll update when I have more.

Randy

david-schmidt, 2018-03-08:
One thing you might do to test the drive is to do a DOS 3.3 bootstrap over serial:
http://adtpro.com/bootstrap.html#Bootstrapping_DOS
With plain old DOS 3.3 in memory, you will be able to format a disk with `INIT HELLO,S6,D1` (to format slot 6, drive 1).  That will give you a known-good formatted disk (or it will report that you have what amounts to a hardware issue).  If it formats OK, then you can try again with ADTPro and see if it can then "see" your drive.

david-schmidt, 2018-05-26:
Closing for now - feel free to reopen when/if you're ready.

closed 2018-05-26 by david-schmidt