ADTPro/adtpro #11: macOS: Only works once before requiring reboot.

thelbane, 2018-03-08:
My configuration:

* macOS Sierra Version 10.12.6
* ADTPro Version 2.0.2
* [RXTX for Mac OS X from jlog.org](http://jlog.org/rxtx-mac.html)
* [Prolific USB Serial Driver PL2303 1.6.1](http://www.prolific.com.tw/US/ShowProduct.aspx?p_id=229&pcid=41)

With this configuration, I'm able to successfully run ADTPro once before having to reboot. If I disconnect/quit ADTPro, run it again and click "Serial," the program hangs and I have to force quit. Any subsequent attempts to run ADTPro result in the icon bouncing perpetually in the dock without showing any UI (which I then need to force quit). In fact, at this point is when my CPU fan turns on and stays on. Activity monitor reports that my CPU is around 10% load, but there doesn't seem to be any particular process responsible for that. Attempting a restart from the Apple menu closes all my windows and shows me a black screen, but the CPU fan just keeps running perpetually until I hard reset.

What's weird is that I've tried [rxtx 2.1-7r2 (stable)](http://rxtx.qbang.org/wiki/index.php/Download) from the official qbang download page, however with that release, ADTPro doesn't seem to recognize any serial ports at all.

Thanks for a great tool. When it works, it works great! Just bummed I'm always having to reboot.

david-schmidt, 2018-03-08:
A couple of things are true on the Mac these days: 
1. You need to run the `adtpro.command` script rather than just merrily clicking on the app.  That will use the embedded RXTX library with the proper permissions applied.  You don't want to be playing with other installable RXTX versions.
2. PL is a terrible, terrible chipset for Mac.  You don't necessarily need to buy one from me, but for your own sanity, please buy one somewhere that uses FTDI.

thelbane, 2018-03-08:
I just ordered an FTDI USB Serial cable, which will be here in a few days. I'll take a look at the startup script later tonight.

Thanks for your help!

thelbane, 2018-03-17:
The new cable + launching with the script seems to be working great! Thank you.

closed 2018-03-17 by thelbane