ADTPro/adtpro #12: https GitHub download links don't work with old browsers

seanm, 2018-03-12:
Since ADTPro is itself related to 'retro computing' I thought you might care to know that it's not possible to download ADTPro from an old browser via GitHub.

You link to https://github.com which (sensibly these days) seems to require modern encryption.  So from Safari on Mac OS X 10.6.8, GitHub is inaccessible.  I'm guessing because that Safari doesn't support TLS 1.1.  Happily, I was able to download Firefox 48 (last that works in 10.6) and could download that way.

Still, you might want to provide some unencrypted http links for anyone trying to access their ancient computers from their merely 'old' computers. :)

david-schmidt, 2018-03-13:
The reality is that TLS 1.0 is going the way of the dodo bird - it will very soon be unsupported by newer browsers, and servers will not be far behind.  Even if I supply (or you simply use) http vs. https in a link, it won't help if GitHub shunts you right back over to a TLS 1.1 connection.  That's not a "me supplying a link" problem - that's an old browser in the new world problem.

Believe me, I'm sympathetic - I use 10.6 for some very good tooling that doesn't exist on newer MacOS myself.  I don't, however, do web surfing with it as it's becoming more and more incompatible.  I think that's the reality we're in going forward - to participate in the "modern" internet, whatever that is, we will always need some new tech that can still bridge back to the older machines.

If, on the other hand, you are suggesting I need to build a less-secure server of my own - donations are gladly accepted. ;-)

seanm, 2018-03-13:
To be clear, I was suggesting that the downloads be mirrored somewhere additional using plain old http. I'm not sure where they could be mirrored. Maybe sourceforge?

I don't much care though. :)

david-schmidt, 2018-03-13:
Oh, that's easy then - they are still mirrored on SourceForge.

iKarith, 2018-03-13:
One of these days, someone who does this kind of thing is going to
reasonably conclude that the way to use these older computers on the modern
Internet will necessarily require a proxy to do so and build one into one
of the Raspberry Pi competitors that makes for a good router.  Might even
do it with one of the older routers that commonly run DD-WRT or similar.

I'd be tempted to suggest this isn't our issue to fix.

Joseph

david-schmidt, 2018-06-01:
Doing some maintenance on issues - I don't think there's anything left to answer on this one, so closing.

closed 2018-06-01 by david-schmidt