ADTPro/adtpro #13: Rom 1 Apple IIGS cannot Bootstrap with Windows 10 PC

jlk001, 2018-03-27:
I'm connecting by Serial via Com 3 from a Windows 10 PC to a Rom 1 Woz iigs.  I'm getting no activity on the iigs end, but the Windows PC successfully recognizes there is something plugged in and acts like it transferred the files for Speediboot after I do the IN#2 and Control A 14B commands.  However the cursor blinks on the iigs and nothing ever happens.  I set the baud rate the same and the slot is designated for a modem in slot 2.  As a matter of fact, if I unplug the USB cable from the PC, the iigs starts beeping and gibberish starts filling up on the screen.  Any suggestions?

RetroFloppySupport, 2018-03-27:
What's the serial connection consist of on the Windows end?  I.e. what chipset is on your (I assume) USB-RS232 adapter?  Did you install the driver for it?  How sure are you it's actually talking on COM3?

jlk001, 2018-03-27:
My Windows 10 PC is connected to the IIGS by a miniDIN-8 to DB-9 serial cable, a Serial DB-9 to USB converter cable based upon the FTDi chipset, slimline null modem adapter.  I bought this kit online: https://www.ebluejay.com/ads/item/5058024
I did successfully install the drivers.  They came with the kit listed above.  I checked it's on COM3 through the Device Manager in Windows.

RetroFloppySupport, 2018-03-27:
Well, it's the cables.  One thing you might try: on the ADTPro server, bring up the File->Serial Configuration dialog box, and tick the "Apple IIc w/Imagewriter Cable" checkbox.  That twiddles the handshaking bits a little bit differently, and may help with your particular cable set.

jlk001, 2018-03-27:
Unfortunately, I tried that option as well.  It's weird, because the IIGS knows that something is plugged in, because if I unplug the USB from the Windows PC, garbage starts to fill the DOS Screen.

jlk001, 2018-03-27:
Could you use ADTPRO with the serial port instead of the Modem Port?

RetroFloppySupport, 2018-03-27:
So that points to something wrong with handshaking (unplugging from the PC).  Time for you to ask your cable vendor for some help.

As for the port - there are some things you can do by hand (bootstrapping the old-fashioned way via text) for the first part of the journey, just to see if the printer port could accept text correctly.  You'd need to configure the port differently since the defaults are very different between printer and modem, and I've not done it successfully before.  But it doesn't really matter beyond debugging if you've got a bad GS port since the main program won't/can't use it anyway.

raicho81, 2018-04-26:
I have similar problem. I try to send ProDOS or speediboot etc. and nothing happens. When I check the  "Apple IIc w/Imagewriter Cable" option it sends some garbage. I've made the cable myself using the pinout scheme from the ADTPro website and I've checked and double checked it several times. Also my USB to serial adapter uses CH340 chip-set and I am using Windows 10. After switching the com port to higher number in the trace I get "SlotNotFoundException" or something similar. Today I bought an FTDI based USB to RS232 adapter and tomorrow I'll try with it because I have some doubts that the one I've tried is crappy.

jlk001, 2018-04-26:
Let me know the results!

raicho81, 2018-04-28:
I've just tried the FTDI based USB to RS232 the first time it didn't work - the progress bar in ADTPro wasn't moving at all after a minute or two. Then I restarted the program and computer and it worked like a charm. I've formatted a floppy disk and there is no problem. Apparently the old USB to RS232 I have is crappy.

closed 2018-04-28 by david-schmidt