ADTPro/adtpro #14: Image transfer aborting while trying to receive ADTPro-2.0.2.DSK on macOS High Sierra

azconapithecus, 2018-04-01:
Hardware:
Apple IIe (unenhanced)
Mid 2015 MacBook Pro 15" running macOS High Sierra
Connected by a null modem serial cable / usb adaptor

After Bootstrapping Speediboot and formatting a blank 5.25 disk, when I try to receive ADTPro-2.0.2.DSK, my Apple gets to around writing Block 00008 of 00280 when the ADTPro app on Mac displays:

 "Image transfer aborted."

On my Apple, the first progress bar will eventually finish at 00040 or around there, wait, then beep and return to the main ADTPro 2.0.2 menu. When I try to repeat these steps, I get the same result. The very first time I tried this, the formatted floppy got to the very last progress bar before the Image transfer aborted. Now it seems to remain at writing Block 00008.

azconapithecus, 2018-04-01:
I ended up trying to format and write to my second disk drive and everything appears to have worked. I was getting a different progress bar this time, much faster than before and without X's being displayed with the writing of each block... which now makes more sense.

So definitely a hardware issue on my first disk ][ drive which is unfortunate for me but does not seem related to the ADTPro. Perhaps a bit more feedback on why the image transfer was aborted might not go amiss?

david-schmidt, 2018-04-01:
Your scenario seems to make sense - it was trying to write, but couldn't, and using up retries, ended up timing out at the host - hence the "aborted" message.  The block 00008 is simply the first batch of data that came in, which was never successful in writing (using your bad drive), as indicated by the XXX display.  Not sure how much more feedback there could be.

closed 2018-04-01 by david-schmidt

azconapithecus, 2018-04-02:
Concerning feedback:

Since I had never witnessed ADTPro in action, the first time seeing the progress bar I simply assumed the "X" in the progress bar meant the program finished writing that block. Even though "X" is commonly used to refer to an error, it is also very frequently used as a confirmation checkmark when filling in boxes and such. I had no reason to assume that the blocks weren't writing correctly as no error type sound or accompanying message was used to indicate failure immediately upon encountering one. Plus the progress bar kept moving on to the next blocks which I wouldn't have assumed would be the case upon encountering an error. Everything seemed fine to me.

After printing the first "X" on the floppy, if a message was shown on the Apple saying "Failure to write block..." or something along those lines, it would make the experience more idiot friendly, and thus be more helpful to someone like me :) Perhaps both the "X" and the message could use some "FLASH" text to tie the two together.

Something similar could be done on the app end on the PC/Mac side. Instead of saying "Image transfer aborted." the message might continue along the lines of "Image transfer aborted. Timed out due to errors writing blocks." or something of that nature.

I say all this for future first-time users who may only have one drive their machine which due to age, they're unaware it's lost its ability to write properly. Luckily I have two and after many Mac restarts and Super Serial Card shuffling attempts, decided to finally try Drive 2. If I hadn't, it's likely I would not have been able to identify the problem myself.

I know this feedback may seem fairly pointless but as a long-time developer myself, I often find interface indicators I thought were stupefyingly obvious need an even bigger mallet-to-the-head treatment.

david-schmidt, 2018-04-02:
I know what you mean - I was working with some beta Apple II hardware that was only ever in failure mode, so when I finally got a version that actually worked, I was amazed that the progress bar should be green instead of red.  If you're working with bad hardware, I'm not sure there's a great way to plan for that.  We tend to assume hardware is good, and don't have a good way to make the experience great if it's not.

Since you've got some bad hardware, I'm all ears as to how you'd make the failure more plain (in the form of a PR, please).