ADTPro/adtpro #19: Virtual drive not working on IIGS and Windows 10

raicho81, 2018-04-29:
I am able to use Speediboot and format disks but my floppy drive is faulty and stopped working at all. I've tried VSDrive + Speediboot and it loads then when I enter "CAT,S2" it says "NO DEVICE CONNECTED". Any ideas?

iKarith, 2018-04-29:
Are you using the printer port for your serial connection by chance?

Joseph

raicho81, 2018-04-29:
No, I am using the serial port. Now I've noticed that while loading it outputs "FOUND SLOT 8" but if I type "CAT S8" I get again "NO DEVICE CONNECTED .

raicho81, 2018-04-29:
Well, it is working actually but the drives are at S1. I saw it when I booted the VSDRIVE disk image from the RAM drive. I think we can close this.

iKarith, 2018-04-29:
That was why I asked if you were using the printer port.  The IIgs has two identical serial ports, printer and modem.  Printer is slot 1, modem is slot 2.  You're plugged into the printer serial port, which is why it's slot 1.

raicho81, 2018-04-29:
I am pretty sure that it is plugged in the leftmost din-8 connector, which has a phone picture besides it but I'll give it a try and plug into the other one

david-schmidt, 2018-04-30:
Virtual serial drives are mapped to slot 1, independent of where your serial comms is coming from.  See:
http://adtpro.com/vsdrive.html

david-schmidt, 2018-04-30:
Depending on exactly how it was plugged in incorrectly, there is usually a chip in the middle of the analog board inside the Disk II that blows - a 74LS125, iirc.  It's rare to fry the controller itself.

iKarith, 2018-04-30:
Did that change David? I need to update some a2cloud documentation I think.

Joseph
Sent via mobile

raicho81, 2018-04-30:
Thanks for the info. I've asked my father too and he said the same - that there is a buffer IC and I've probably blown it's inputs with 12V.

david-schmidt, 2018-04-30:
@iKarith It didn't change recently - the doc I pointed to above talks about how the search for a slot with two free drives works, and it's independent of the physical comms port.

iKarith, 2018-04-30:
Recently could be 2015 in our case. ;)

Joseph
Sent via mobile

david-schmidt, 2018-06-01:
Doing some maintenance on issues - I don't think there's anything left to answer on this one, so closing.

closed 2018-06-01 by david-schmidt