ADTPro/adtpro #20: Can't get ADTPro on Windows 10 to talk to the IIGS

xthree, 2018-05-02:
My brain hurts. A lot. 

Set up: 

Client
Apple IIGS Woz Edition with ROM Version 01 (heard it's buggy)

Connections
 [8-Pin Mini DIN to DB-9](https://www.amazon.com/gp/product/B019DE6G0G/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1)
[DB9 Null modem adapter](https://www.amazon.com/gp/product/B075XHWVSJ/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1)
[USB to Serial Cord with FTDI chip](https://www.amazon.com/gp/product/B004WL7MIQ/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1) (No clue if the FTDI chip is legit)

Server
Windows 10 with Java 8 (does Java version matter?)
I have installed the driver from the [Gearmo website](http://www.gearmo.com/downloads/)

I have gone into the IIGS control panel and made sure the items were all default. Slot 2 is set to modem.

When I boot the IIGS and go to the prompt, I type IN#2, I press control+A (this turning the cursor into a question mark is not reliable sometimes it works usually never the first time though), then 14B. Then I press return.

When I attempt speediboot, absolutely nothing happens on the IIGS, but the progress bar on ADTPro moves and says file was sent in 15 seconds.  I have not changed the baud or any other settings on ADTPro, but I did notice the modem port settings on the IIGS displays 1200 baud, I doubt that's relevant since it is shown on your screenshot on your site, but thought I'd bring it up.

Other maybe relevant data, I tried this all before with a different cable (Prolific, didn't know it was bad until afterwards). But I found that if I took the null modem adapter off the set up, random data was being written to my screen. ADTPro wasn't even running. Just random garbage all over. Maybe that indicates my modem port isn't busted? I attempted the same set up (no null modem) with the new cable with the FTDI chip and no garbage data was displayed on the screen.

Anyways, I'm at a loss, should this work with windows 10? I'm very close to installing Win 7 onto a computer and trying it from there.

xthree, 2018-05-02:
Is it possible my pinout could be wrong even though I am definitely getting SOME good data transferred?

david-schmidt, 2018-05-02:
That's exactly what's wrong: handshaking.  As evidenced by it sorta-working when you use the ImageWriter tickbox (which helps with that particular cable's brain-dead wiring).

xthree, 2018-05-02:
So I cut open my cable, and there were only 3 wires inside. I just ordered the one from retrofloppy, like I should have done originally. 

Thank you very much for pointing me in the right direction. I'll report back once that cable comes in.

david-schmidt, 2018-05-03:
Sounds great.  Thanks!

xthree, 2018-05-24:
Got it working and was so excited that I forgot to report back. All is working great now with the new cable. Thank you so much for the quick replies. I am having a blast with my IIGS!

closed 2018-05-24 by xthree

david-schmidt, 2018-05-24:
Fantastic news.  Enjoy!