ADTPro/adtpro #21: Syntax Errors

Tucsonbroker, 2018-05-07:
When I try bootstrapping via a ProDos speedboot, hitting reset as soon as the computer boots and hitting IN#2 then CTRL-A 15B, the Computer gets text sent to it, but can not use it and causes a SYNTAX ERROR.   The Prodos ADTPro disk boots, but I get the unable to contact remote server error.

SSC is in Slot 2, has the correct DIP switch settings, and the Arrow is pointing to MODEM.  I've also tried a straight through cable with the Arrow pointing to TERMINAL.   Same result of garbled text on the screen.

Here is the weirdest part.   Of the dozens of times I have attempted to get it to work, it has worked twice.   Nothing has changed via the card or the cabling.  I can not get it to replicate the environment in which it did work.

I suspect the Super Serial Card is bad.   When I first booted with the card and tried the IN#2 command the screen filled with random characters and I had to repeatedly hit CTRL-A to get the prompt.  Sometimes PR#2 works and other times it doesn't.  I have reseated the chips on the card and verified that all pins have connectivity.  So something must be wrong in the ROM.  I have it connected to a Null Modem, then to a USB adapter.  Purchased both cables from RetroFloppy.  Then it goes to an Apple USB to USB-C adapter.  I've also tested via an older mac which uses USB and have had the same results.  Tried different Slots in the computer.

Any other thoughts or things I should try?  I have a replacement SSC coming this week.

Notes:
Franklin Ace 1000
MacBook Pro, latest OSX.
SSC in Slot 2

david-schmidt, 2018-05-07:
It doesn't strike me as a hardware problem in the SSC, or in the cabling pipeline in general (though a USB-C adapter in the middle is a new wrinkle for me).  My first suggestion is to pay close attention to the items in the pick-list on the ADTPro server's serial configuration dialog box.  When you bring it up with your USB adapter unplugged, note what's in the list (probably some bluetooth garbage).  Then, close it, plug in your USB adapter, and bring up that serial configuration dialog box again.  One or two items will appear that weren't there before; those are the ones that pertain specifically to your USB adapter.  Be sure you are using one of those (it should not matter which of the two similar ones you choose).

Once you've got that chosen, fire up the Franklin, hit reset, and do the `IN#2` command.  It should not be spewing garbage at you at that point since you haven't sent anything yet.  If it does, there may indeed be something wrong with the SSC.  If all is still well, proceed with `CTRL-A 14B` and see where that gets you.  If things are still OK, then it's time to send Speediboot.

Tucsonbroker, 2018-05-07:
Thanks for the quick reply.   As part of the troubleshooting process, verifying I was using the correct item in the drop-down menu.   The Mac sees the DB9-USB converter when I plug it in.  

I thought it might have been the USB to USB-C dongle so I used another Mac that has standard USB ports and I got the same issue.  

Obviously the Mac and Franklin are talking together because it is receiving characters.   Both times Speediboot worked it was after some crash/issue happened on the Franklin while accessing the SSC.   Once when I hit PR#2 the screen went black and I rebooted and it worked.   Then rebooted and it didn’t work.  The second time was right after I physically removed the SSC and reseated the jumper on the card.   Worked right away and when i tried a second time it failed.   

Also of note:    The first time it did work I was able to transfer 3 images to 5 1/4” floppies.   Only one of the disks booted.

david-schmidt, 2018-05-07:
Awesome, thanks for those clarifications.  So a couple more things: one, you mention `PR#2` - there is no scenario in which you should be `PR#`ing to your card.  `IN#` only.  The Franklin Ace 500 note does not apply to you.

Second, and probably most important - it sounds to me like the SSC card really only works when it's freshly powered up.  No amount of rebooting or resetting seems to help it.  What results do you get if you power-cycle the whole machine and try after a fresh power-up?

Tucsonbroker, 2018-05-07:
Power cycling doesn’t seem to help.   I even let the Franklin sit for a minute to make sure everything clears from the ROM.   I really do think it’s the SSC given the weird behavior it has exhibited.   I’ll uninstall ADTPro and do a fresh install with the adapter plugged in just to rule that out.   Regardless I have another SSC coming so I’ll report back results.   Fingers crossed.

david-schmidt, 2018-05-07:
I hate to blame the hardware, as it seldom is at fault - but sometimes bad things happen to good chips!

Tucsonbroker, 2018-05-09:
Replacement SSC fired up and works perfectly.   Bummed that I wasted a lot of time diagnosing this, but happy that it works.   Thanks for your help.  Case closed.

closed 2018-05-09 by david-schmidt