ADTPro/adtpro #22: how to move a disk image i download

scooter87, 2018-05-18:
I'm using a mac book pro using os sierra 10.12.6 i got adtpro and it works but when i go to put my disk image into the disk folder in adtpro it will not let me move it into the folder and i don't know what I'm doing wrong and i haven't been able to find any info for this

david-schmidt, 2018-05-21:
This sounds like a question about moving files in the MacOS finder rather than something about ADTPro - am I reading that right?  Can you be any more specific with what you're seeing?

scooter87, 2018-05-21:
yes it will not let me movie files to the disk folder i getting my files from apple2online

david-schmidt, 2018-05-21:
Ok, but what is preventing you from dragging them in?  Is it a folder permission problem?

scooter87, 2018-05-21:
im not for sure how to fix that problem

david-schmidt, 2018-05-24:
Right, but when you opened the .dmg, did you copy the ADTPro folder to the Applications folder as instructed, or are you running it right from the .dmg?

scooter87, 2018-05-25:
wow thank you sorry about this its working fine now

closed 2018-05-25 by david-schmidt