ADTPro/adtpro #25: serial port issue on windows 7

ficelle3, 2018-05-26:
I tried connecting my apple//e to my windows 7 computer with ADTpro but audio didn't work,it seemed that the apple// wasn't able to send data to the computer so I bought a usb to serial adapter, which shows up in the device manager as a serial port, but ADTpro doesn't see it, when I try to select the port, ADTpro doesn't list any,and if I press 0K, ADTpro tells me that there is no rxtx library or serial port.can someone help me please

david-schmidt, 2018-05-26:
It sounds like you got an appropriate driver installed if you can see it in Device Manager.  Does it tell you which COM port it inhabits?  That will be important later.

If you really have a serial port but ADTPro can't see it, the most likely cause is a 32-bit Java installed on a 64-bit Windows.  If you bring up a command prompt and enter `java -version` that should give us some clues.

ficelle3, 2018-05-26:
the coverter was originaly on port COM5 but I tried changing it to COM1 and I've tried using java 32bit and 64bit version 1.8.0_171

david-schmidt, 2018-05-26:
Definitely don't try to change the COM port it comes in on.  And you want to match bit-width of Java with your OS - very likely 64-bit in both cases.

ficelle3, 2018-05-27:
I changed the port back to COM5 and unistalled the version of java I had to install the newest 64 bits version but ADTpro still gives me the same error.

david-schmidt, 2018-05-27:
From a command prompt - what does `java -version` say?

philgoetz, 2018-08-23:
Are you sure the COM port you're seeing in the device manager is the right one? And that it has a driver installed? If you bought a cheap Chinese USB-to-serial connector, it probably uses the CH341 chip, meaning you need to get a driver for it from www.wch.cn . Or better, buy one not made in China.

david-schmidt, 2018-09-21:
Feel free to come back with an issue again - closing for now.

closed 2018-09-21 by david-schmidt