ADTPro/adtpro #29: MicroDrive and ADTPro Woes.

willgonz, 2018-06-03:
I have a https://wiki.reactivemicro.com/MicroDrive/Turbo in my Apple IIGS in Slot 1.  Whenever I use ADTpro 2.0.2. the MicroDrive won't boot.  I have ADT Pro 1.1.6. and it works fine.  If I try to load ADT Pro Ethernet 2.0.2 it hangs my MicroDrive and won't load.  Trying a different Slot maybe 7.

david-schmidt, 2018-06-04:
When ADTPro starts up and doesn't have a configuration saved, it does probe the slots to see what cards it can find.  If you save a definition for it, it should skip the probe and leave the MicroDrive unconfused.  See: http://adtpro.com/configethernet.html
Specifically, the "Save Configuration" part.

david-schmidt, 2018-06-27:
I've now got a MicroDrive and will try some similar scenarios.  Can you explain what you mean by "hangs my MicroDrive and won't load"? Do you mean you can't run ADTPro from your MicroDrive, or do you mean that after running ADTPro and exit you can't reboot and start from the MD, or...?  Maybe some steps that clearly demonstrate the problem will help.

willgonz, 2018-06-28:
Yup when the MicroDrive is in Slot 1 and I go to use ADTPro 2.0.2 it will not catalog my hard drives that are on the Compact Flash Card and it locks up ADTPro.  If I control open apple Reset then the compact flash won't boot.  I have to power completely down in order for the card to work again.  I moved the Compact Flash Card to Slot 7 and ADTPro will work fine with it.

david-schmidt, 2018-06-29:
Got it.  Even though you said you used slot 1 in your first message... it never occurred to me to put one there. :-)  Will go against my better judgement and put it in slot 1 and see what happens.

david-schmidt, 2018-09-11:
Ok, the problem has been been found and worked around.  The situation needs to be the following:

1. MicroDrive/Turbo with IIgs firmware
2. ADTPro Ethernet version that automatically scans for Uthernet cards (i.e. 2.x)
3. MD/T needs to be in a slot **before** the Uthernet card

The problem does not manifest if the MD/T has IIe firmware, nor does it happen if the MD/T is in a slot **after** the Uthernet.  The IIgs firmware on the MD/T is unable to recover from the slot scan for the Uthernet; the IIe version is immune to the scan.

So the ADTPro code now tiptoes around the IIgs firmware if it is found during the slot scan; and a real bug was fixed where ADTPro scans for cards even if a configuration file exists.  So look for the next release after 2.0.2...

david-schmidt, 2018-09-14:
Released 2.0.3, and... closed.

closed 2018-09-14 by david-schmidt