ADTPro/adtpro #34: JavaApplicationStub approach is way outdated...

inindev, 2018-08-06:
I am running OSX High Sierra 10.13.6 and am seeing flaky behavior with JavaApplicationStub. Even though I have the Oracle JDK v8 installed and JAVA_HOME declared, launching the ADTPro app requests the legacy Apple Java v6 runtime to be installed. Against my better judgement, I went ahead and installed the legacy runtime. While the app does now launch, it is very flaky and crashes when I click the "Disconnect" button. I also cannot see that I am able to generate a trace log file as the Apple JVM is launching the app in some strange location (due to quarantine?).

However, I do see the "ADTPro.command" script you have included and it properly uses my Oracle Java v8 runtime, is _perfectly_ stable, and generates log files in the correct location. Yeah!

I guess I have two questions:

1) Why are you naming "adtpro.sh" "ADTPro.command" on the Mac? Doing so caused me to not find it for a long time. I am unfamiliar with what a ".command" file is. I would propose just leaving it "adtpro.sh" to make its purpose more obvious to Mac users.

2) What are the ramifications of ditching JavaApplicationStub? My testing shows that deleting ADTPro-2.0.2.app (and everything under it) and replacing it with "ADTPro-2.0.2.app/Contents/MacOS/ADTPro" where the ADTPro is a shell script which contains:

    #!/bin/sh

    export ADTPRO_HOME="`dirname \"$0\"`"
    cd "$ADTPRO_HOME"
    export ADTPRO_HOME="`pwd`/"

    "$ADTPRO_HOME"adtpro.sh

I have changed my system to the above and I am very pleased with the result. I thought I would socialize these changes here for feedback before submitting a PR.

david-schmidt, 2018-08-06:
> Why are you naming "adtpro.sh" "ADTPro.command" on the Mac? Doing so caused me to not find it for a long time. I am unfamiliar with what a ".command" file is. I would propose just leaving it "adtpro.sh" to make its purpose more obvious to Mac users.

I see you're new to Mac OSX. ;-)  In The Beginning (tm), the Mac Finder (crica 10.1-10.3) would not recognize a `.sh` as a double-clickable entity.  If you named it `.command`, however, it all of a sudden became double-clickable.  The tricky thing is... we have users that go all the way back to 10.3, and they remind me every time I accidentally move too far forward (such as using Java 8).

> What are the ramifications of ditching JavaApplicationStub? 
None, now that we have a way to invoke the shell script - and it also solves the problem of draconian permissions problems introduced in 10.13 (or maybe it was 10.12).

inindev, 2018-08-06:
>I see you're new to Mac OSX. ;-) In The Beginning (tm), the Mac Finder (crica 10.1-10.3) would not recognize a .sh as a double-clickable entity. If you named it .command, however, it all of a sudden became double-clickable.

I skipped the powerpc years, after the Apple IIe I did not use Apple until OSX.

My preference would be to just use a Raspberry PI anyway, but I am unable to find the arm RXTX library referenced here:

```
  if [ -f /usr/bin/raspi-config ]; then
    export RXTXLIB=lib/rxtx/%RXTX_VERSION%/arm
    ADTPRO_EXTRA_JAVA_PARMS="-Dgnu.io.rxtx.SerialPorts=/dev/ttyUSB0:/dev/ttyAMA0"
```

Is there a ADTPro discussion board where I can ask questions outside of the github issues room?

david-schmidt, 2018-08-06:
Most general ADTPro discussion happens in usenet space (comp.sys.apple2).  There are lots of places with slightly different audiences, but the most relevantly technically inclined will be there, including myself.  The Pi side of things is inhabited by a slightly different set of folks than me, but they'll also be able to help with the Pi-specific prereqs.  https://github.com/dschmenk/apple2pi is a good place to be looking for Pi goodness as well.

inindev, 2018-08-09:
If you want to give the new install approach we discussed above a try, I have staged it in my fork:

https://github.com/inindev/adtpro/releases/tag/v_2_0_3

If it passes muster I can create a PR for your repo.

--John

david-schmidt, 2018-08-09:
Looks good on 10.13 - which solves the permissions problem getting access to devices.

closed 2018-08-13 by inindev