ADTPro/adtpro #40: Investigate ProDOS 2.4.2?

inindev, 2018-08-13:
I noticed adtpro uses ProDOS 2.4.1, does it make sense to update the base image to 2.4.2 for the next release?

https://github.com/ProDOS-8/Info/releases

david-schmidt, 2018-08-13:
Indeed it does.

david-schmidt, 2018-08-13:
I know and trust Oliver Schmidt, but I'd want to learn more about how John Brooks is moving his code from his usual distribution method (Call-A.P.P.L.E.) over to github:
http://www.callapple.org/vintage-apple-computers/apple-ii/announcing-prodos-2-4-2-for-all-apple-ii-computers/

inindev, 2018-08-13:
It looks like it is located there too:
http://www.callapple.org/soft/ap2/system/ProDOS_2_4_2.dsk

david-schmidt, 2018-09-10:
Integrated with this commit:
https://github.com/ADTPro/adtpro/commit/0c8ba17f0daec09496ec81d618bd85d8a62ae97a

closed 2018-09-10 by david-schmidt