ADTPro/adtpro #41: Consider web-page based server

inexorabletash, 2018-08-16:
This could be done as a separate project that just shares the protocol, but kicking off conversation here.

Currently, ADTPro requires installing a Java runtime, which these days (2018) is a significant burden or impossible on some platforms. 

As the web advances in capability, it is possible to implement at least some of the server options on the web, so that an ADTPro user needs only an up-to-date web browser.

* The Audio protocol could be supported via [WebAudio](https://webaudio.github.io/web-audio-api/) for audio signal generation and processing
* The Serial protocol could be supported via a new Web Serial API being discussed [here](https://discourse.wicg.io/t/serial-api-moving-from-web-of-sensors-wg-to-web-incubator-cg/2940)

If interested, comments on the Serial proposal would be welcome - we're looking for developers who could take advantage of it and could be early adopters of an implementation.

david-schmidt, 2018-08-17:
I love the idea!  When will serial be ready? ;-)

inexorabletash, 2019-08-30:
The Serial API is now available in Chrome Canary builds (and stable, but might be an older iteration) behind a flag (`--enable-experimental-web-platform-features` on the command line, "Experimental Web Platform features" in chrome://flags)

Very draft spec is at: http://wicg.github.io/serial/

We're interested in getting examples of use of the API and feedback to ensure the design is correct so we can advance it along the standardization process, ship it in Chrome stable without the flag, and hopefully get implementations in other browsers.

I haven't had a chance to actually play with it with my Apple II. (I have a not yet tested port of my old vnIIc video-streaming-over-serial ported to the web using this and other modern web APIs.)

david-schmidt, 2019-09-03:
This is starting to sound very promising.

inexorabletash, 2019-10-17:
Downloaded a 80MB JRE to run ADTPro... nope, need to download the 180MB JDK which will blow up to 300MB on disk... yeah, let's make this happen.

inexorabletash, 2020-02-23:
The serial API is now available in Chrome 80 as an "origin trial"; you have to use a header or meta tag to enable the feature for a few milestones, but it is close to shipping. 

https://github.com/a2stuff/vnIIc has my example code which bootstraps nearly identically to ADTPro.

inexorabletash, 2021-03-17:
The Web Serial API is now available as of Chrome 89 and Edge 89 and later. 

API Documentation: https://developer.mozilla.org/en-US/docs/Web/API/Serial
API Specification: http://wicg.github.io/serial/