ADTPro/adtpro #42: Question: hardware handshaking

handyandy802, 2018-08-18:
Is there a reason hardware handshaking is not used?

david-schmidt, 2018-08-19:
Not a good one!  What have you got in mind?

handyandy802, 2018-08-19:
Hi David,
Back in the day I used hardware handshaking cable with Proterm and modems over voice phone lines without issues on a Laser128ex or null modem cables between computers for file transfers up to 19.2k bps.  I'm wondering if hardware handshaking isn't necessary anymore.  Is it redundant to use rts/cts or dtr/dts?  Is it a cycle-saving feature?  What's the bad reason?

david-schmidt, 2018-08-19:
One (bad) reason is history.  HW handshaking has never been used in any of the projects ADTPro descended from, so it's not been seen as a priority.

One (good) reason is that each type of device implements hardware handshaking differently - or not at all well.  The SSC is diffferent from the IIc is different from the Laser 128 family is different from the SCC.  It's another layer to manage for no useful gain.

handyandy802, 2018-08-22:
Hi David,
From what I gather ADTPro runs in a polling mode.  I'm wondering if HW handshaking (other than IRQ) is even necessary.  Reason I'm asking is that I'm working on another project and considering using just IRQs for serial port handshaking and not even bothering with dsr/dtr or cts/rts.  My feeling is that the extra handshaking is a holdover from voice line modems.  My understanding is that all serial ports either on send register empty or receive register full there's an IRQ.  Check status register and go...

Cheers!Andy

david-schmidt, 2018-08-22:
Right, it is totally polling.  There's no IRQ servicing at all.  It's all based on register registering availability, and servicing in a loop fast enough to not miss a character.  The protocol itself basically dictates half-duplex communications, so polling works well.

handyandy802, 2018-08-22:
Thanks David,
You've satisfied my curiousity.  I am a bit old-school when it comes to tele-communication when back in the day everyone was experimenting with 6551 interfacing (late 80s-early 90s).
Cheers!Andy

closed 2018-08-23 by david-schmidt