ADTPro/adtpro #46: Need parameters for serial connection

philgoetz, 2018-08-23:
I need to know what values ADTPro expects for these parameters on its serial connection:
Data bits: 8 or 7?
Parity: One of {None, Even, Odd, Mark, Space}
Stop bits: {1, 1.5, 2}
Flow control: {None, Xon/Xoff, Hardware}

david-schmidt, 2018-08-23:
It's not given because it's not relevant.  All comms settings are done automatically by the host and client except for which port to use (and in some cases) what speed to use.  None of those should ever need to change from the defaults (except, again, the port).  If you're trying to, that means something else is wrong.

philgoetz, 2018-08-23:
I have to set them on the device driver for my RS232-to-USB device.

david-schmidt, 2018-08-23:
They'll be overridden by the rxtx communications layer, so it doesn't matter.  If they somehow demand you set them, just use the defaults for every serial connection there ever was since 1986 - 8,N,1,N.

closed 2018-08-23 by david-schmidt