ADTPro/adtpro #47: What will the ADTPro server do if the ADTPro disk-image directory is not accessible due to privileges?

philgoetz, 2018-08-23:
Since my ADTPro fails only when trying to read the ADTPro-2.0.2\disks directory, I have a horrible suspicion that I may have one of those Windows access control problems.

What would the ADTPro server do  if the user who ran it somehow doesn't have permission to read the disks directory?

It appears that all users have read access to that directory, but for all I know, I might be having that obscure "you have read access to this file, but I'm not letting you read it anyway because you downloaded it from another machine" problem.

david-schmidt, 2018-08-23:
> What would the ADTPro server do if the user who ran it somehow doesn't have permission to read the disks directory?

I imagine it would return the errors the operating system reported - which might manifest themselves in ways you describe.

> I notice that when I try "Change working directory", the Java dialog shows no files in the directory, even though there are files in that directory and it is set to show Files of Type "All Files". This happens in all other directories on the machine, though, so I guess it is really set to show only folders.

That's normal - it's restricting you to just navigating among directories.

So tell me what it means when you say "Since my ADTPro fails only when trying to read the ADTPro-2.0.2\disks directory"?  Does that mean that the client won't show directory contents when you do a `(D)IR`, or something else?

philgoetz, 2018-08-24:
It means that I get a host timeout error after about 2 seconds any time I do a (D)IR, and after I type a filename on a (S)end or (R)eceive.

david-schmidt, 2018-08-27:
Ok.  I think that's just back to the fact that you don't have a good host-Apple II connection as we discussed earlier.