ADTPro/adtpro #53: Problem transfering files with parenthesis in filename

Mark-Five, 2018-10-10:
Platform : Win10 x64
Java : 8/181 x64 (latest)
Using ADTPro release 2.0.3 on apple //e with SSC
Filename : SHERWOOD FOREST (1982)(PHOENIX SOFTWARE).dsk

Content of trace window : 
Exception in thread "Thread-2" java.util.regex.PatternSyntaxException: Unclosed group near index 40
SHERWOOD FOREST (1982)(PHOENIX SOFTWAR..
        at java.util.regex.Pattern.error(Unknown Source)
        at java.util.regex.Pattern.accept(Unknown Source)
        at java.util.regex.Pattern.group0(Unknown Source)
        at java.util.regex.Pattern.sequence(Unknown Source)
        at java.util.regex.Pattern.expr(Unknown Source)
        at java.util.regex.Pattern.compile(Unknown Source)
        at java.util.regex.Pattern.<init>(Unknown Source)
        at java.util.regex.Pattern.compile(Unknown Source)
        at java.util.regex.Pattern.matches(Unknown Source)
        at java.lang.String.matches(Unknown Source)
        at org.adtpro.gui.Gui$1.accept(Unknown Source)
        at java.io.File.list(Unknown Source)
        at org.adtpro.gui.Gui.getFiles(Unknown Source)
        at org.adtpro.CommsThread.sendDirectoryWide(Unknown Source)
        at org.adtpro.CommsThread.dispatchCommand(Unknown Source)
        at org.adtpro.CommsThread.commandLoop(Unknown Source)
        at org.adtpro.CommsThread.run(Unknown Source)

Renaming the file (removal of parenthesis) solves the problem.

david-schmidt, 2018-10-10:
Good catch.  Is this because you had selected the file via wildcard on the client end, or did you type it all out?

Mark-Five, 2018-10-10:
Pressed R for receive, then RETURN.
Then selected the file using arrow keys and RETURN again.
Funny thing is that it's starting something on the serial port.
After that, I have to disconnect/reconnect (on the PC software) to regain access to ADTPro.

david-schmidt, 2018-10-10:
Got it, this is via the arrow/return interface.  Another thing is also true - if the file (name) is actually longer than the screen, it will simply fail to find the file (since the file name is passed as-is-viewed to the server).

Mark-Five, 2018-10-10:
Yep. Also noticed this one but thought the filename was just too long.
I can help on the 6502 part but I suck at java.

david-schmidt, 2018-10-10:
The Java side is trivial.  But more interesting might be to send a
page+offset index to the file rather than screen-scraping as it does
today.  That would solve both problems.

david-schmidt, 2018-10-10:
Notes: 
In `hostfns.asm`: `NIBPCNT`holds the current page number, `COL_SAV` is current vertical position on screen (variable name re-use is unfortunate in this case); it will be 3-indexed (first row is actually 4th on screen due to header).

Protocol extension to [Query File Size](http://adtpro.com/protocol.html#Query_File_Size_Packet) and [Image Get](http://adtpro.com/protocol.html#Image_Get_Packet) will include somehow encoding the page and row to send to the host.  Reply from the host can remain unchanged.

david-schmidt, 2018-10-11:
Merged - will be part of next release!

closed 2018-10-11 by david-schmidt

jblang, 2020-09-12:
This error still occurs on ADTPro 2.0.3.  

```
ADTPro Server version 2.0.3

Exception in thread "Thread-2" java.util.regex.PatternSyntaxException: Unclosed group near index 40
KING'S QUEST III (4AM AND SAN INC CRAC..
        at java.util.regex.Pattern.error(Pattern.java:1969)
        at java.util.regex.Pattern.accept(Pattern.java:1819)
        at java.util.regex.Pattern.group0(Pattern.java:2922)
        at java.util.regex.Pattern.sequence(Pattern.java:2065)
        at java.util.regex.Pattern.expr(Pattern.java:2010)
        at java.util.regex.Pattern.compile(Pattern.java:1702)
        at java.util.regex.Pattern.<init>(Pattern.java:1352)
        at java.util.regex.Pattern.compile(Pattern.java:1028)
        at java.util.regex.Pattern.matches(Pattern.java:1133)
        at java.lang.String.matches(String.java:2121)
        at org.adtpro.gui.Gui$1.accept(Unknown Source)
        at java.io.File.list(File.java:1161)
        at org.adtpro.gui.Gui.getFiles(Unknown Source)
        at org.adtpro.CommsThread.sendDirectoryWide(Unknown Source)
        at org.adtpro.CommsThread.dispatchCommand(Unknown Source)
        at org.adtpro.CommsThread.commandLoop(Unknown Source)
        at org.adtpro.CommsThread.run(Unknown Source)
```

The filename I get the error on is "King's Quest III (4am and san inc crack) disk 3A.dsk". The client reports host timeout and the server then has to be restarted in order to successfully transfer anything again.  Removing the parentheses fixes the problem.   Interestingly "Frogger (usotsuki prodos port).dsk" does not cause the same error. I believe it only happens if the character following the open paren is a number.  Looks like Java is interpreting the combination of open parentheses and a number as a capture group number. Any parentheses probably need to be escaped in the filename using \\.

david-schmidt, 2020-11-24:
Patience... the "next" release hasn't been released yet.

jblang, 2020-11-24:
Well in fairness, the comment I was responding to is 2 years old at this point.