ADTPro/adtpro #56: OSX Server Always Returns Host Timeout After 1 Second

iethereumny, 2018-10-14:
Bootstrapping works fine for both ProDos and ESDos. I've tried every setting possible and with ADTPro loaded on the Apple II. (R)quest and (D)ir commands always returns "HOST TIMEOUT". Only takes 1 second to return so there must be communication handshake problems. Again boot strapping works fine so not a cable issue.

-macOS High Sierra version 10.13.6
-ADTPro-2.0.3.dmg
-Apple ii with SSC 2

iethereumny, 2018-10-14:
The first part of speediboot works however it stops on the CLI screen

david-schmidt, 2018-10-14:
Ok, if Speediboot won't work all the way, then there's something wrong with your connection.  It tests bi-directional communications as part of what it does.

So possible problems so far are :
  1. The null modem cable may be suspect (if you didn't get it from RetroFloppy)
  2. The SSC card may be able to receive, but not send

iethereumny, 2018-10-14:
-Cables are from retrofloppy
-SSC points downward to "terminal", tried "modem" and bootstrapping stopped working, SW1: 1001111  SW2: 1101100

iethereumny, 2018-10-14:
I've tried the FTDI drivers too. I think next I'll try to find a windows computer

iethereumny, 2018-10-15:
I will also try the apple /// null modem cable from RetroFloppy. It's not super clear from the adtpro website which cable is required for USB.

My setup:
Apple ][ with SSC2 connected to OSX USB 

Already tried: "USB to RS-232 Adapter" and "DB25M to DE9F Adapter – Straight Through" with arrow down on SSC2 (Terminal)

Going to try: "USB to RS-232 Adapter" and "Apple ///, Super Serial null modem" with arrow up on SSC2 (Modem)

iethereumny, 2018-10-15:
It's possible the SSC2 is bad although it's NOS with box. I may order another one to try. Sounds like you're pretty confident the OSX server is working properly.

david-schmidt, 2018-10-15:
I've been doing this long enough to have seen one of everything fail along the long path of serial communications.  A failure at any point can and does lead to a bad connection.  Without adequate spares to test which is at fault, we're all relegated to guessing.  The cables leave RetroFloppy having been tested live with ADTPro, so I am confident they are the least likely candidate.  Hardware is next-least-likely - it doesn't often fail, but one side of the SSC can go bad (i.e. it can send or receive, but not the opposite).  Software is in the middle somewhere - drivers can be mis-installed or conflicting, Java levels can be mismatched between bit-width of the machine and the distribution, and so on.  The list is endless.  Your instinct to go with a Windows box is a good one - it eliminates a whole set of variables if it is found to work.

iethereumny, 2018-10-15:
I'll report back. My setup is pretty typical I would imagine so probably worth sorting out: New MacBook Pro (apple fans remain apple fans) with latest OSX connected to Apple ][ with SSC2.

david-schmidt, 2018-10-15:
I have one of everything to test with, so I'm confident there isn't a problem with working infrastructure.  There's nothing inherent with your setup that precludes it from working.

david-schmidt, 2018-10-16:
One thing came to mind as I was running your symptoms through my head: it has happened that the pins between the USB adapter and the serial cable didn't quite meet.  On one occasion, I had had to "deflect" the male pins off by a few degrees (I made a kind of herringbone pattern of the pins: 
`" \ / \ / "` is an exaggeration, but you get the idea) in order to make sure every pin was contacting its mate in the cable.

iethereumny, 2018-10-17:
Happy to report the problem was indeed a bad serial card, exactly as you had guessed. Lesson learned, "opened boxed" items from the 1980s are sometimes faulty items that were set aside. I just finished copying Castle Wolfenstein and it worked flawlessly. Looking forward to copying more images!

My working setup:
Apple ][ with SSC2 connected to OSX 10.13.6 High Sierra via USB
SSC2 settings: arrow down(Terminal) SW1: 1001111 SW2: 1101100
"USB to RS-232 Adapter" from retro floppy
"DB25M to DE9F Adapter – Straight Through" from retro floppy 
FTDI driver from FTDI website version 2.4.2 
By the way, High Sierra now includes an FTDI drivers by default however I removed it from /System/Library/Extensions and used the FTDI driver instead while troubleshooting. It might work with the default FTDI driver in High Sierra

david-schmidt, 2018-10-17:
Fantastic - glad you got that sorted out.  Too bad about the SSC #1 - no fun having dud hardware.  But it's not all that rare for one side of the SSC to go bad and not the other.  There's just no good way to distinguish where along the path the signal is getting lost without some known-good spares, unfortunately.  And good to know about the drivers, too - I had been using them in a MacBook Pro without realizing I hadn't installed the driver myself!  So I can confirm the Apple one actually works.  My guess is it's the same one as from the factory.

closed 2018-10-17 by david-schmidt