ADTPro/adtpro #57: Apple IIes don't receive any data from different ADTPro servers

rexs1234, 2018-10-26:
I'm having problems using ADTPro via null-modem serial link.  I have installed it on both MacOS and Windows 10.  I also have two Apple IIes.  One enhanced platinum and one non-enhanced.  They each have their own serial cards.  I configure the cards with the block pointing up and the switches as shown in ADT's site.  I also do the IN# of the slot and get "Apple SSC" and enter the correct number.  When I run the software on both the Mac and Win10 it starts to send the data and the light on my data cable blinks showing it is sending data.  The progress on the server says it's sening data.  I get no response from the Apples and nothing appears on the screen after starting the send.  The server says the data sent and completes.   I believe both cards are working cards as I use them as printer cards with no issues.

I'm using the Apple2OnlineStore cable kit with FTDi chipsets.  Here is the link for the kit I have:
https://www.ebluejay.com/ads/item/5557255

Any suggestions?

david-schmidt, 2018-10-26:
The settings look correct.  I'm assuming you're using slot 2 (good to start with normal defaults).  Have you contacted your seller?

rexs1234, 2018-10-27:
Yes, I've actually tried several slots including #1, #2, and #4.  I did reach out to the cable seller and they are working with me to help isolate if it might be a cable issue.  They are a great seller I've used before.  I just posted the link to show what setup I was using.  

David, do you have any other suggestions I can try?  Are there any commands on the Apple IIe side I can execute or areas of memory I can check to see if data is actually reaching the card or what it is doing with it?

david-schmidt, 2018-10-27:
During bootstrapping operations, characters (good or bad) will appear on the screen because of the `IN#2` and `<Ctrl-A>14B` commands you issue.  But that assumes about 42 other things are all aligned and working - a few of which are:

1. Your cable is a crossover (i.e. null modem) and you've got the jumper block pointing down; you might try flipping the direction, just for fun - but it's not super-likely that's the problem
2. Your USB cable has a good FTDI driver on your PC, and you know for sure which COM port it populates when you plug it in (a good way to check is to see what new COM port appears in the hardware manager when you plug it in)
3. You're picking the correct COM slot in the ADTPro server when you are sending the Speediboot commands
4. Your SSC is actually viable (they do go bad; it's a nice extra that you have two to test with)
5. You're sending the SSC commands with upper-case characters and aren't somehow mis-typing the `<Ctrl-A>14B` bit

rexs1234, 2018-10-27:
I've actually tried the jumper block in both positions.  However, the ADTPro page seems to state the opposite of you in that regard:

To connect with a straight-through cable, set the Super Serial card's block to point downwards 
toward the word "Terminal," as in this picture:
To connect with a null-modem cable, set the Super Serial card's block to point upwards toward the word "Modem," as in this picture:

That text and pictures are from: http://adtpro.sourceforge.net/connectionsserial.html

I guess I'm uncertain exactly which way it should point for sure based on this.  I have a null modem cable.  Having the jumper in the correct position for sure would make debugging the other parts much easier.

david-schmidt, 2018-10-27:
I was winging it.  The web page is correct.  But ultimately that's not the issue, since you've tried it both ways.

drkenb, 2018-11-12:
We wound up replacing the FTDi & cable with an FTDi board and its now working fine for him.  We also came across a second customer who is having a problem with the FTDi cable kit working on his SSC card.  Not sure why all our other SSC card customers had no problems but these two are.  Going to look into the revision SSC cards they have.  If you have any other ideas, would love to hear them!

david-schmidt, 2018-11-12:
It really can be anything at all.  I've seen it help to deflect the pins a couple of degrees just so they rub against the connector a little differently.  I'll close this issue since it's not particular to ADTPro.

closed 2018-11-12 by david-schmidt