ADTPro/adtpro #58: adtpro crashes on apple iie immediately after 800G command

smitty97, 2018-10-28:
I picked up an enhanced IIe with duodrive but have no bootable disks. 
Not having any luck with audio bootstrapping.  Nothing on asciiexpress.net works either, from iphone or pc laptop audio.  EsDOS seems to load ok, but trying INIT DISKNAME gives syntax error. 

Sending ProDOS works, then 2nd part loading ADTpro crashes immediately at address $8412.  

screen showing crash: https://i.imgur.com/tuUZj5M.jpg

help!  could there be a problem with the iie?

david-schmidt, 2018-10-29:
It does sound like a hardware problem, most likely.  EsDOS is easiest on the system because it doesn't require all 64k to be functioning.  A syntax error specifically is an odd symptom; can you do anything else such as CATALOG once it's running?

Not that I expect it to work any better, but does Speediboot do anything different for you?

smitty97, 2018-10-29:
thanks for the tip; I came across ctrl-closedapple-reset hardware test and sure enough ram chip #5 fails.  :(   You can ignore this ticket.    Nice work on this software btw! hope to get it going on some new (old) hardware soon.

closed 2018-10-29 by smitty97