ADTPro/adtpro #59: "Bootstrapping" menu item grey on windows 10

andrewfinch, 2018-10-31:
I have the server running with JRE1.8.0 on windows 10 and it thinks it is successfully connected to COM4 where I have an FTDI usb serial interface installed and visible to the system, but the "Bootstrapping" menu item is grey.

I also tried it on my macbook pro running Mojave using the same usb serial device and it connected and I was able to go through the bootstrapping transfer procedure, but I was never able to receive anything on my Apple IIc - which is why I was trying it on my windows machine. These may be unrelated issues.

Any assistance would be greatly apreciated, thanks

david-schmidt, 2018-10-31:
Hi, Andrew -

The Bootstrapping menu is generally disabled when the server is either disconnected or connected via Ethernet (since it doesn't support bootstrapping).  If you've pushed either the serial or the audio button, it should become available.  Part of "pushing the serial button" includes selecting an appropriate COM port (on the PC) or /dev (on the Mac).

andrewfinch, 2018-10-31:
Yes I have selected serial and the com port For the usb serial device and it displays “connected” but still can’t use the bootstrapping menu. Maybe it’s not actually connected? Is there something I should mess with in JRE? Thanks

david-schmidt, 2018-10-31:
And it always stays disabled?  Even if you hit the disconnect button and then the serial button?  Do you see any errors on the console window (that hides behind the UI - you have to right-click on it in the windows start bar)?

andrewfinch, 2018-11-01:
Yes it always stays disabled.  There are errors:

Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: com/webcodepro/applecommander/storage/Disk
        at org.adtpro.utilities.VDiskPersister.refreshDisks(Unknown Source)
        at org.adtpro.utilities.VDiskPersister.<init>(Unknown Source)
        at org.adtpro.utilities.VDiskPersister.allocateSingleton(Unknown Source)
        at org.adtpro.utilities.VDiskPersister.getSingleton(Unknown Source)
        at org.adtpro.CommsThread.<init>(Unknown Source)
        at org.adtpro.gui.Gui.startComms(Unknown Source)
        at org.adtpro.gui.Gui.actionPerformed(Unknown Source)
        at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
        at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
        at javax.swing.JToggleButton$ToggleButtonModel.setPressed(Unknown Source)
        at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
        at java.awt.Component.processMouseEvent(Unknown Source)
        at javax.swing.JComponent.processMouseEvent(Unknown Source)
        at java.awt.Component.processEvent(Unknown Source)
        at java.awt.Container.processEvent(Unknown Source)
        at java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Window.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
Caused by: java.lang.ClassNotFoundException: com.webcodepro.applecommander.storage.Disk
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        ... 43 more

david-schmidt, 2018-11-01:
Great, this is helpful.  How about restarting the server, and observe what messages come up, and when.  So my questions now are: 

Do any messages come up immediately after start?

Does a new message come up when you hit the serial button?

Are you starting ADTPro with the batch file (not just clicking on the .jar, right)?

andrewfinch, 2018-11-01:
Aparently I was launching the command script instead of the batch file... I really thought I tried both of these... but it works now so I guess it was user error.

I am still not able to communicate with my IIc but I have a usb serial adapter which has a female db9 and a gender changer to make it fit the null modem cable which maybe is causing a problem so I’ll try swapping that out, unless you have any other suggestions.

Thanks a lot for your help.

david-schmidt, 2018-11-01:
I have an axiom after all these years... "it's the cable."

closed 2018-11-01 by andrewfinch