ADTPro/adtpro #62: ADTPro 2.0.2 opening on macOS Mojave

Woogetybop, 2019-01-23:
Unable to run ADTPro 2.0.2 java application with latest Java 8 on Mojave 14.2

david-schmidt, 2019-01-23:
I assume you mean 2.0.3?  Changing you title to reflect that.

I understand the `ADTPro.command` doesn't work for you.  2.0.3 is supposed to help alleviate the need for that by introducing changes in the app icon, so do please give that a whirl too.

At a terminal, what does `java -version` say to you?

Woogetybop, 2019-01-24:
Yes my error. Version 2.0.3 not 2.0.2
Version 2.0.3 will not run by app or command.
I’m using version 2.0.2 right now just fine.

As for java version I checked at the terminal and that’s my problem. 32 bit java 6 JDK was active. I installed the JDK java 8 package and that fixed the issue. I had installed Java 8 JRE runtime, but 64bit JDK was required.

Thank you again for all your time and effort on what I consider a remarkable program. 

Best Regards,

Tod Sanders
woogetybop@gmail.com

david-schmidt, 2019-01-24:
That's a pretty common problem on the PC side of the fence, but I didn't even realize it was possible to mess up the bit-width of your JRE on Mac.  Wonders never cease!  Glad you got that straightened out.

closed 2019-01-24 by david-schmidt