ADTPro/adtpro #64: Adtpro 2.0.3 does not support the transmission of 40-track 160K disks.

qwj5708, 2019-02-12:
Adtpro 2.0.3 does not support the transmission of 40-track 160K disks. It uploads 40-track 160K disks to PC. It only generates 140 K mirror files for 0-34 tracks, while data for 35-40 tracks are ignored. I hope that future versions of ADTPRO will consider compatibility with 40-track 160K disks.

david-schmidt, 2019-02-12:
Where are some examples of 160k disks to work with?  Are there any images floating around?  What DOS knows how to manipulate 40-track disks?

qwj5708, 2019-02-14:
I attach several example images. The source disks of these images are 160K disks with 40 tracks. After transmission with ADTPRO, only 0-34 tracks of data are retained, while 35-40 tracks of data are ignored. There are many examples. I hope ADTPRO can be compatible with 40-track 160K disk format. Thank you.

david-schmidt, 2019-02-14:
I'm not seeing any attachments.  Again, can you tell me what DOS writes these disks?

qwj5708, 2019-02-14:
My three attachments were emailed to "David Schmidt" <notifications@github.com> yesterday (February 13, 2019). Didn't you receive them? My 40-track 160K floppy disk is written in with the improved DOS 3.3. Those improved copying software can read and write 40-track 160K disk. I will send you another copy software of track 40 by email today. Please check it.

david-schmidt, 2019-02-14:
That email won't get to me.  You can just drag and drop files directly into the message window to upload them right here.

One more time - what is the DOS that writes these?  Where can I download this DOS?

qwj5708, 2019-02-14:
[Disk Mirror Example.zip](https://github.com/ADTPro/adtpro/files/2865670/Disk.Mirror.Example.zip)

david-schmidt, 2019-02-14:
How did you virtualize the 160k images you have?

qwj5708, 2019-02-15:
This is how I made the 160K image: file type (you can see files with CATALOG) I used COPY] [PLUS copy one file to another empty disk 1, until 140 K (0-34 track is basically full), and then copy the remaining files to empty disk 2. Then ADTPRO is uploaded to PC to generate 1.DSK and 2.DSK. Then COPY [PLUS] is run in the APPLEWIN simulator, and the files in 2.DSK and 1.DSK are copied into a 160K empty DSK. Empty 160K.DSK can be downloaded at https://www.gryphel.com/c/minivmac/extras/blanks/.

Non-file type (CATALOG can't see files) disks are copied to analog floppy drives with arbitrary track copy software to form two DSK files, and then merged into a 160K DSK in PC.

These methods are cumbersome, and it's also cumbersome to turn 160K mirrors into disks. Therefore, it is urgently hoped that ADTPRO can support 40-track 160K disks and mirrors.

patrickbst, 2019-02-15:
For what it is worth, the actual hard limit on disk size in DOS 3.3. was 400K, though I am sure this was done through one patch into DOS or another.  The one I am most familiar with is if you had a Corvus OmniDrive/OmniNet, it allowed 400K volumes (in fact, it did a really really cool thing, that you could make a 2000K volume for DOS which it would divide into 5 400K volumes that you addresses by using the V1...V5 option that no one else used).  There is a patch someone else made for DOS to support 2 400K disks on the 3.5 drive.  The 400K hard limit has to do with the MAX that Tracks can be and sectors can be (not sure the nums).  I think most of the time some Hardware that support bigger dsk sizes (like hard drives) included these patch versions of DOS that allowed them to address the bigger sizes.   If you want a reference for this, look for the 3.5 disk support for DOS, that dev may be around.  Of course qwj may be able to help here too.  If he wants ADTPRO  to talk to 160K disks, I assume he must have one of those physical drives he is trying to read from/write too (or had one in the past) and the OS disks for that drive should be around somewhere.  Also look on asimov in hardware section for disk drives.  I seem to recall someone made a 160K drive that plugged into apple (forget who).  Also for some reason I keep thinking that maybe DiversiDos supported the extra space DOS volumes.

qwj5708, 2019-02-16:
Thank you for your attention. I have more than 240 disks with all kinds of software. Of them, 160K of 40 tracks accounts for 50%. It has been more than 30 years now. Some disks can hardly be read out. In order to save these disks, I have used many ways to convert the disks into mirror files and save them on the PC. Among them, ADTPRO is the best tool to convert disk to mirror, but ADTPRO has a good effect on 140 K disk of 35 tracks, while for 160K disk of 40 tracks, the conversion will abandon 35-40 tracks, resulting in incomplete mirror content. This is the main reason why I am here to ask the help of David Schmidt, the author of ADTPRO software. It is hoped that the next version of ADTPRO will add support for 40-track 160K disks. In order to completely rescue all my disks.

jartester, 2019-07-15:
Rana ELITE series of drives use multiple DOS formats up to 600K (double sided 40 tracks). The RWTS is patched and additional utilities are supplied.

david-schmidt, 2019-07-15:
All quite interesting - but the bottom line is that ADTPro uses ProDOS to interact with drives.  Unless/until a ProDOS driver exists for these devices, this isn't something I'd entertain adding support for.  Feel free to try to change my mind with a PR, though!

closed 2019-07-15 by david-schmidt