ADTPro/adtpro #67: Macintosh plus serial port

Tommy51389, 2019-02-27:
Hello again!

I saw your previous comment, and I was wondering what cable I would need to connect from my Macintosh plus to my laptop?

I was thinking the DIN 8 serial modem port? I’m not entirely sure though. I assume that the receiving end would plug into the db25/parallel/printer port on the laptop side correct?

Thanks,

david-schmidt, 2019-02-27:
What software are you planning to run across the serial wire?  There's not an application of ADTPro between a Mac Plus and a PC, so I'm not sure how this applies.

david-schmidt, 2019-02-27:
It sounds like maybe you're wanting the Mac to be the client (as if it were an Apple II) and then feed it disk images via serial from the PC?  That's not possible with the current software.

Tommy51389, 2019-02-27:
Yes I was referring to the Mac as the client and feeding disk images via serial from the pc. 

I understand now, thank you for you help!

closed 2019-02-27 by Tommy51389