ADTPro/adtpro #68: Having trouble running on Windows 10 with AdoptOpenJDK 8

edolnx, 2019-02-28:
The default adtpro.sh doesn't seem to work on Windows 10 Build 1809 when using AdoptOpenJDK 8 x64:
```
C:\ADTPro-2.0.3>adtpro.bat
The system cannot find the file -Xms128m.
```
I tried removing the `start /min` part of the command, and `java.exe` spits back a very unhelpful usage without any error as to why.

I also tried `'C:\Program Files\AdoptOpenJDK\jdk-8.0.202.08\bin\java.exe' -jar .\lib\ADTPro-2.0.3.jar;.\lib\rxtx\ch-rxtx-2.2-20081207-win-x64\RXTXcomm.jar;.\lib\AppleCommander\AppleCommander-1.3.5.13-ac.jar org.adtpro.ADTpro` which appears to work, but doesn't find any serial ports (I have a PL2303 USB/Serial adapter showing as COM4 in Device Manager).

Any thoughts? I would like to avoid the Oracle Java packages moving forward for all the usual reasons...

david-schmidt, 2019-03-01:
It looks like the normal command line switches for Java (from Oracle and IBM) are not being heeded by your vendor.  -Xms128m is a directive to allocate a particular amount of memory to the initial heap size:
https://stackoverflow.com/questions/4667483/how-is-the-default-java-heap-size-determined

If your vendor isn't using switches like everyone else expects, I'm not in a great position to help.  But that error message was pretty clear.  

The RXTX serial library has a native hardware component that needs to be accessible to Java - it works fine in other distributions, so how to make the dll work somewhere else is going to be an exercise left to you, I'm afraid.  The batch file as shipped has pathing set up so that the library will be visible to the JVM when it spins up.

edolnx, 2019-03-01:
I double checked, and the Java heap command line options do work in AdoptOpenJDK:
`'C:\Program Files\AdoptOpenJDK\jdk-8.0.202.08\bin\java.exe' -Xms128m -jar .\lib\ADTPro-2.0.3.jar;.\lib\rxtx\ch-rxtx-2.2-20081207-win-x64\RXTXcomm.jar;.\lib\AppleCommander\AppleCommander-1.3.5.13-ac.jar org.adtpro.ADTpro` works, but again - no serial ports listed and no information on the console as to why. I think part of the problem is that the `start /min` command in the batch script seems to have been broken by a recent Windows 10 update. The serial port problem, I agree, may be related to the Java distribution. I will file a bug with them.

closed 2019-03-01 by edolnx