ADTPro/adtpro #69: ADTPro Stops Working on Format Screen

stweedle, 2019-03-09:
New ADTPro514 disk boots and displays boot menu but stops working on format screen.  I have put in a new unformatted floppy and have also left in the ADTPro disk.  None of the format screen options work including "esc to Cancel"

Apple IIe with Super Serial Card in slot C1 and floppy control card in slot C6.  Heads cleaned but no change.

Odd that it will boot from the floppy but can no longer see floppy after ADTPro loaded into memory.  

Any ideas?

david-schmidt, 2019-03-09:
The format function goes through and interrogated each slot for a ProDOS device.  When you to the format screen, does it give any hints what it’s finding?  It’ll print out one line per slot that it finds.  If not, I suspect something is wrong hardware-wise with slot 7.  Unless... you’ve got something plugged into slot 7 you didn’t mention.

stweedle, 2019-03-10:
Thanks for the quick reply.
Super Serial in slot C1, the floppy controller in slot C6 - should it be in C7?  All other slots are empty.
Once the format screen comes up I get nothing - no disk seek, no lines filled-in, and none of the options like (R) or ESC work. 
Yet the ADTPro initially boots fine on floppy 1 and I can open the "config" and "dir" options from the menu - have not tried the others, well, other than format.  
Related (maybe): I have the Super Serial card dip switches set as recommended, jumper block pointed to "terminal,"  RetroFloppy cables, Windows 10 32-bit, Java 32-bit, installed FTDI CDM21228_Setup.exe for drivers.    

Thanks...

stweedle, 2019-03-11:
Tried moving Super Serial from C1 to C2 and floppy controller from c6 to c7.  Cleaned contacts of each card and air cleaned slots.  Same results.
I do note that on the initial boot screen  "select your contact type" that the "IIe" is not listed (attached screen cap).  Do I have the correct version of ADTPro?

thanks
[image]

david-schmidt, 2019-03-11:
You've got the right one - SSC is what you're using.

I'm assuming the `V` for volumes screen behaves the same way?

So something is going wrong with the slot query, having nothing to do with your cabling.  It's all about the interaction of ProDOS with your hardware.  What you might do now is do a plain DOS (not ProDOS) bootstrap, and then with DOS see if you can format a disk.

stweedle, 2019-03-13:
Latest...
1. You're right: the "volume screen" hangs, too

2. You're right: Disk 1 could read but not write. I removed Disk 1 and replaced it with Disk 2.

3. ADTPro still had the same discovery problems. I have tried the purchased ADTPro disk, ADTPro server with serial and audio cables, Windows 10 and Java 32-bit and 64-bit. 

4. I was able to format and successfully transfer programs over audio cable from ASCIIExpress.net. With a transferred copy of DOS 3.3 I was able to boot and format about 10 floppies and then transfer a game.

5. From applefritter.com found Apple floppy calibration pages which looks very educational but probably beyond my current skill level: appleii-box.de/H084_1_AppleIIDiskService1.htm     

(Note: Above links scanned with VirusTotal.com URL checker)

Thanks for your help and pointing me in the right direction. Why I still can't get ADTPro to work is a mystery but at least now I have some bootable floppies and a way to create more. My next step will be to open Disk 1 with hope to see something obvious - ha! 

I am new to vintage Apple hardware and finding the adventure fun, so far, with great support and information in forums such this and applefritter.com. So, thanks to all.

david-schmidt, 2019-03-13:
The thing that is going wrong is still hardware related, isolated to the Apple II.  It's possible that memory is bad, a slot is bad, or both.  The thing that ProDOS does that plain DOS does not is it exercises all of the memory in the Apple IIe (and also scans for hardware differently).  So issues on the Apple are more often caught with ProDOS running than other things.  If you can use ASCIIExpress to create you a ProDOS disk like the ProDOS user disk that is capable of formatting, it would be interesting to see if that would work.  Otherwise, any diagnostic type program would also be helpful.

stweedle, 2019-03-15:
**The adventure continues...**
Transferred "MECC Computer Inspector" from ASCIIExpress, booted and it reports: 
	"auxiliary memory needs repair, error 04 at 1084A"

Removed the card, cleaned the contacts and reseated it with same error results. In addition to memory, this card has the monitor port and the nearest I can tell (from eBay) it is an "Apple IIe Extended 80-Column RGB Card" (I have a color monitor).  

eBay looks to be a minefield with card prices all over (e.g. <= $449) and most reasonably priced cards are "untested."  

Even so, maybe the card is ok but the slot and/or motherboard bad.  Bonus bad news, my color monitor is beginning to distort in the upper left corner. Bad monitor/card/slot/motherboard. Who knows. Jeesh! 

So, looks like an Apple hobbyist must have access to available/inexpensive parts to troubleshoot and those parts are mostly out of reach, price wise or are a gamble . 

Thanks - you've been a great help. Apple hobbyists are such a great group, willing to offer help and software tools.

Not sure what I'll do next, look for parts on Craigslist and continue on eBay I guess.  Maybe find replacement memory chips as they are in sockets.  

Thanks again.

stweedle, 2019-03-31:
**Update - good News (so far)**
So took a chance and ordered replacement memory chips (MT4264-20) for the auxiliary card.  I replaced all 8 chips as I did not know which one bad and the originals were what seemed a mix of manufacturers.

**Bottom-line:** Everything now works! ADTPro communicates over the serial line, formats and writes disks, even the monitor seems to have lost its distortion.  Almost too good to be true so hope I don't jinx it.  

ADTPro works like a champ as I transferred half dozen games to newly formatted disks. 

So, there you have it, was a hardware problems as you predicted - so another big **Thanks Again!**

david-schmidt, 2019-04-01:
Awesome - nothing quite so satisfying as fixing your own stuff!

closed 2019-04-01 by david-schmidt