ADTPro/adtpro #70: Cannot run ADTPro on Windows 10, 98, XP or OS X El Capitan on four machines

afsec92, 2019-04-15:
I am having problems starting ADTPro on four different machines, Windows 10, 98, Xp, and OS X El Capitan. On each machine the JAVA has been verified and tested with another .jar program that work perfectly. I am summing that JAVA is ok. My suspicion is that adt.bat or adt.cmd converted to adt.command script is the issue. I do not see any instructions about the script other then "rem" statements. I have tried JAVA jre-6u45-windows-i586 without any success. Untalled jre-6u45-windows-i586 and then installed 8 Update 201. 
I have FTPI drivers installed.

david-schmidt, 2019-04-15:
I'm not sure you're going to have a lot of luck on either Win 98 or XP, but 10 and OSX should be fine.  What can you tell about the messages you're getting in the console when you start the batch or .command files?

afsec92, 2019-04-15:
Here are the messages from the terminal.

Last login: Mon Apr 15 16:44:15 on ttys000
Scotts-MacBook-Pro:~ ssimon$ /Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command ; exit;
/Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command: line 1: @ECHO: command not found
/Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command: line 2: REM: command not found
/Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command: line 3: REM: command not found
/Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command: line 4: REM: command not found
/Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command: line 5: REM: command not found
/Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command: line 6: REM: command not found
/Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command: line 7: REM: command not found
/Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command: line 8: REM: command not found
/Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command: line 9: syntax error near unexpected token `('
/Users/ssimon/Desktop/ADTPro-2.0.33/adtpro.command: line 9: `REM e.g. uncomment (remove the "@REM" from in front) and customize'
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...none found.

[Process completed]

david-schmidt, 2019-04-15:
You don’t want to run a .BAT file from OSX - what does running the .command do for you?

afsec92, 2019-04-15:
I was running the adt.command on my MB. Those were the results from the terminal.

david-schmidt, 2019-04-15:
Ok, that doesn't quite make sense - the `REM` statements are a batch file construct.  Did you download the `ADTPro-2.0.3.dmg` from the [downloads page](https://github.com/ADTPro/adtpro/releases)?

afsec92, 2019-04-16:
Evidently I did not have the latest version from GITHub. I downloaded ADTPro 2.0.3 and installed the .dmg file. Next I had a message to install JAVA Development kit. I download from Oracle's website the JDK file 8_201. Bingo! ADTpro 2.0.3 is working on my MAC Book Pro El Capitan. I like the fact that the ADTPro application is residing in the application folder.

I hope other individuals read this solution.

Thank you David Schmidt for your fast response and assistance.

closed 2019-04-16 by afsec92