ADTPro/adtpro #75: no serial ports at all

urzu117, 2019-07-10:
Running w10 x64, with an apple IIc with original ROM. I cannot get ADTpro 2.0.3 to find any of my PCIe serial ports at all. I've loaded RXTX correctly, with the correct version for 64 bit java. it's a WCH PCIe dual serial card, if that helps at all.

david-schmidt, 2019-07-10:
So you have one of these?
https://www.asix.com.tw/products.php?op=pItemdetail&PItemID=122;74;110&PLine=74

Have you installed the driver - looks like "Windows 10/8.x 64-bit Driver   V3.18.1.0" is the right one?

What COMx ports do you see when you run (is it hardware manager? can't quite remember)?

urzu117, 2019-07-13:
It's this one https://www.amazon.ca/gp/product/B00B5L9P36/
and as for the driver, W10 couldn't find one, so I ended up using the drivers included on CD. 

As for COM ports, I got 3 and 4 by default, though I can reassign them as needed.

[image]
this is the driver version

urzu117, 2019-07-13:
Also, I should mention, I am using a homemade DB9 to IIc serial null modem cable, and I've wired it according to the article on adtpro's main website. Shouldn't matter though, since the program doesn't see the COM ports regardless of whether or not the cable is attached.

urzu117, 2019-07-13:
Update: through the power of dark magic I do not understand, the COM ports are suddenly showing up in ADTpro 2.0.3 (aka I cold booted my PC up for the first time in 3 days)

urzu117, 2019-07-13:
another update: can't do speediboot, but can do the seperate ProDOS and ADTpro bootup, but cannot get the disk directory, HOST TIME OUT error happens all the time. Using the other serial port doesn't seem to make a difference, so I suspect this pcie card is just bad. I'll try to get a USB DB9 adapter tomorrow.

schmidtd, 2019-07-13:
> another update: can't do speediboot, but can do the seperate ProDOS and ADTpro bootup, but ...
That means you've got one-way communications working, but not two-way.  From the Apple II perspective, you can receive, but not send.  99.9% of the time, this is a cable issue.

david-schmidt, 2019-07-13:
>another update: can't do speediboot, but can do the seperate ProDOS and ADTpro bootup, but ...

That means you've got one-way communications working, but not two-way. From the Apple II perspective, you can receive, but not send. 99.9% of the time, this is a cable issue.

urzu117, 2019-07-14:
Well, you seem to be psychic. Despite me checking the night before with a multi-meter for continuity between each connection, Pin 2 on the DIN socket was not connected to anything. Upon further inspection, it appears as though the solder was too weak at the wire and pin connection and the wire broke free. 5 mins of repair, and it's working. I was able to do speediboot and then write ProDOS 2.4.2 to disk. Thank you for your help.

closed 2019-07-14 by urzu117

david-schmidt, 2019-07-14:
I wouldn't claim clairvoyance, but experience does count for a lot.  After 13 years of helping folks worldwide with serial connections in ADTPro, some patterns do emerge.