ADTPro/adtpro #76: A null modem cable issue or OSX Mojave (10.14.5)

fpsurgeon, 2019-07-21:
First of all, thanks for creating this amazing piece of software to begin with!

Here's my problem, though. I can't get ADTPro to properly sync communications with the IIGS from my MacPro. I get a very fast send of Speediboot from the server app (sent in 9 seconds) but just gibberish on the IIGS end. It's as if the data is being sent too fast or otherwise not syncing. I *assume* the transfer should start by dropping the GS into the monitor with a "CALL-151" and then blasting in some machine code, but I just get gibberish.

The USB-to-Serial cable is using an FTDI chipset. The null modem cable is one I wired myself according to the specs on the ADTPro webpage (double and triple checked with my multimeter, too). To me, the gibberish seems like a data rate or flow control issue, but I'm no expert, especially with how quickly the server completes the Speediboot send.

Mac is running MacOSX Mojave 10.14.5. GS is a ROM01 (with all defaults in the GS's control panel). ADTPro version is 2.0.3 launched as .command. 

Hope you can help me.

[image]

david-schmidt, 2019-07-21:
The full instructions for doing a bootstrap from bare metal are here:
http://adtpro.com/bootstrap.html#Starting_from_bare_metal

You didn't describe much about your process, and I see you did do IN#2, so that's good - but if you're making assumptions about Call -151, I think you need to read through the instructions first.  Pay particular attention to step 4, which has two sub-steps that are very important.

fpsurgeon, 2019-07-22:
Yes, thanks. I've read those and thought I've been following them exactly. I've attached the screenshots of my sequence on the server end and on the IIGS end I indeed do a "IN#2" to get input reading from the modem port followed by a <ctrl>-A (to get the blinking "?") and then key in "14B" to select the data rate. But when I then hit "OK" on the server end, all I see is the gibberish of random characters blast across the screen. Hitting <return> multiple times as suggested in the instructions also makes no difference. Note that the string of characters is different each time also.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

fpsurgeon, 2019-07-22:
BTW, what I meant about **CALL -151** is that I suspected that the text stream sent from the ADTPro server might start with such a character sequence to begin setting up the IIGS... not that _I_ had to key that in. Makes me wonder if the DE-9 to Mini DIN-8 is where the problem lies. Are you sure the ones from RetroFloppy follow the same wiring diagram on your webpage or is there another way to wire it? 

BTW, if I order a DB9 to Mini DIN8 from RetroFloppy just to make sure, what's the shipping time (if you know)?

david-schmidt, 2019-07-22:
>... I indeed do a "**IN#2**" to get input reading from the modem port followed by a *Control-A* (to get the blinking "?") and then key in "*14B*" to select the data rate.
Ok, that was my question - it feels a lot like the `Ctrl-A 14B` dance was missing.  Thanks for clarifying.  Now, you're not booting a disk before you start all this, are you?

fpsurgeon, 2019-07-22:
Booting? No, no. Of course not. When the IIGS starts up I immediately hit `ctrl-reset` to get the DOS prompt. Since I have a couple of IIGSes (both ROM 01) I've tried it on both and gotten the same results. I mention this because one of the GSes has an old hard drive in it and the other one is just bare bones plus the 1MG memory expansion card. 

Apparently, GitHub doesn't allow me to post a video clip of the action to show you what's occurring, but I think you get the idea. Makes me wonder _strongly_ about the wiring of the null modem cable, but it shouldn't be _that_ sensitive. It is wired per your diagram, though. Any other detail you think I'm missing before I try plunking down the cash on a RetroFloppy cable (which would be interesting to test the pinout on it versus the one I cobbled together).

fpsurgeon, 2019-07-22:
Well, here's my workaround for posting video...as an animated GIF. Obviously no sound of me clacking at the keys. It's just me keying in the sequence and then starting the transfer...of sorts. Don't know if this is helpful:

[image]

david-schmidt, 2019-07-22:
This does strike me more as a speed issue rather than a wiring one.  But that could also manifest itself in the driver of the USB-RS232 adapter you've got.  Whatever the issue is, it's common to the null modem cable, the USB-RS232 adapter, or its driver.

fpsurgeon, 2019-07-22:
Yes. That's what I'm leaning toward. The driver is the FTDI driver from FTDI's website (driver v2.4.2 for Mac OS X). The serial-to-USB cable is this one: [https://www.amazon.com/gp/product/B01MCZM6OM](url)

At my office, I have an old leftover Windows box running XP (or maybe Win7) but it has an actual old school 9-pin serial port. To narrow down the possibilities, I'll see if I can fire it up and directly connect the GS to it via the null modem cable with no USB converter. That'll rule out the effects of the USB driver. If you don't mind keeping this thread open, I'll try it out tonight when I get home and report back.

david-schmidt, 2019-07-23:
No good reason to suspect an FTDI-based device; they've proven solid.  The Windows machine sounds like a good way to further triangulate the problem.

fpsurgeon, 2019-07-23:
I was unable to use the old Windows box last night (dead power supply), but I'll see if I can grab another one that has a true RS-232 connector and test it out. Will let you know.

fpsurgeon, 2019-07-23:
BTW, do you know roughly how long it should take for the ADTPro server to send the Speediboot code to the GS? Nine seconds seems a bit quick at 9600 baud unless Speediboot is tiny (like 8K).

david-schmidt, 2019-07-23:
If you mean the first part, the grub bootstrap loader - it's 3515 bytes long.  It's the only part that goes over at 9600 baud, and is subjected to textual input routines.  That loader knows how to shift various hardware into its maximum speed setting, and that's it's only goal in life - the minimum code to get the machine talking at its maximum speed.  It should be under 10 seconds, anyway.

fpsurgeon, 2019-07-24:
Okay, that helps quite a bit. Still haven't had a chance to try a new Windows machine with built-in RS-232 port, but I hope to report back this week with the results.

fpsurgeon, 2019-07-27:
Just tried a connection between the Windows PC with a true RS-232 port, my IIGS, and a null modem cable wired per your specs and I get the exact same problem: a random string of gibberish with Speediboot in ADTPro 2.0.3. The random string of gibberish is _different_ each time. 

So this would argue against:

- Mac vs. PC. 
- USB-to-Serial adapter vs. true RS-232 serial port.

This seems to narrow the problem down to either:

- the home made null modem (although it's exactly to spec per the ADTPro website...and after all, some kind of characters are moving).
- the version of Java (Version 8 Update 221, Release date July 16, 2019)
- the current version of ADTPro (unlikely if you've actually transferred data with that).

I'm kind of leaning toward Java being the problem. Any new thoughts?

david-schmidt, 2019-07-27:
...or there's a problem with the IIgs' serial port.  But you mention you have two GS machines, and both behave the same way.  That leaves you and all of the cabling between machines as the common denominator. :-)

I generally keep up with Java versions, and v8 hasn't proven problematic anywhere else.  They don't generally do anything that mucks with serial communications.

fpsurgeon, 2019-07-27:
Then it's probably the null modem cable. :-/ 

I wonder if, as a test, I could use the _printer_ port (Slot 1) instead. I seem to recall that it was somewhat different, though, than a straight serial port. Would a `IN#1` work instead? 

Might also try just a plain ol' terminal app like PuTTY on the PC side to see if that sheds some insight. I'm not sure why my null modem cable shouldn't work (or would produce those problems unless there's some other interference happening on the cable).

david-schmidt, 2019-07-27:
I've not had a lot of luck running on port 1 on the GS, even trying to set everything like a modem port in the control panel.  And any port autodetection that happens will always find and use port 2 before ever selecting port 1.

fpsurgeon, 2019-07-27:
Thanks. BTW, I'll be on vacation for the next week so I won't have a chance to tinker further for a while, but I'm determined to get this thing working. Please keep this thread open until then.