ADTPro/adtpro #77: ADTPro not sending Win7/64

diskman2019, 2019-07-26:
Running Windows 7/64 Pro.  Installed the latest JAVA 64 (jre-8u221-windows-x64.exe) and ATDPro 2.0.3.  Logged in as Administrator on my Lenovo.  Using a REAL COM1.  Also tried a TrendNet USB>Serial cable.  Using a purchased ADTPro cable.  Tested with meter and its wired correctly with no bad connections or high resistance.

ADTPro loads and sees COM1 just fine.  I type the commands on my Apple //c and it never receives anything.  I know the cable is good as I can communicate between my Lenovo (Win7) and the //c (ZTerm) just fine.  Can even send files.  

When the //c IS waiting for the data from the PC via COM1, I can fire up the terminal program on the PC and I can see my typing on the //c so I know it's getting data.

Why isn't ADTPro not sending data to COM1?  Ideas?  I have tried 6 different PCs and a Mac.  Win 10, 7 and 8.  Various JAVA installs and versions to no avail.

diskman2019, 2019-07-26:
I *THINK* I have it covered now.  I was able to sucessfully send data to a LINK MC5 Terminal at work.  Someone on Facebook mentioned setting the port to 2400 instead of the 9600 default.  At least I KNOW I have a computer that will send data.  Now it's up to the Apple to get the info at this point.

david-schmidt, 2019-07-27:
You definitely don't want to set the port speed to 2400.  If the default speeds don't work, there is something else wrong; typically cabling or driver software.