ADTPro/adtpro #78: No network detected under Windows 10/x64/build 1803 & Java 1.8 build 221

amishrabbit, 2019-07-26:
I'm using ADTPro 2.0.3 on an Apple IIe that has an ethernet card. 

When run on OSX client laptops. the IIe can see the server and the server app shows the local IP address in the title bar of the application. Tested send/receive and it works perfectly.

However, using a Windows laptop, the server application does not display the IP address in the title bar, and does not seem to be serving, well, anything. The IIe cannot see the server when I manually enter the correct IP address into the Confi(G) interface.

I have had to modify the command line to the following in order to get Java to run correctly. In the absence of these changes, Java errors out with the message that "Windows cannot find '-Xms128m'" (screenshot attached) presumably because more recent Java for Windows requires a slightly different way of using the CLI flags than it used to.

start "C:\Program Files\Java\jre1.8.0_221\bin\javaw.exe -Xms128m -Xmx256m -jar" "%ADTPRO_HOME%\lib\ADTPro-2.0.3.jar" %ADTPRO_EXTRA_JAVA_PARMS% -cp;"%ADTPRO_HOME%lib\rxtx\%RXTX_PATH%\RXTXcomm.jar";"%ADTPRO_HOME%lib\AppleCommander\AppleCommander-1.3.5.13-ac.jar" org.adtpro.ADTPro %*

I am running the batch script, with this modified start line, from an elevated/Administrator command prompt window. Windows does not do the normal prompting for firewall passthrough that I would expect, so I suspect that Java is not actually opening a listening socket, but I have no evidence of that.

I've been looking around for solutions to this particular problem, specifically involving using network for transfer, and I've come up empty handed. Do you have any suggested remedies?

amishrabbit, 2019-07-26:
[image]

amishrabbit, 2019-07-26:
[image]

david-schmidt, 2019-07-27:
I'd like you to go back to the default 2.0.3 batch script and post what shows in the command prompt window that is "behind" the server GUI.  That should provide some clues as to what is going on.

amishrabbit, 2019-07-27:
Hi David. Thanks for the quick reply.

So, I tried running the original version of the batch file but I realized it doesn't display any useful data with the /min flag set after the START command, so I only modified it slightly to get rid of that, and this is the output.

[image]

This instance also ran without me configuring the MY_JAVA_HOME variable, and as a consequence, it runs from a different location than the other one did (I didn't even realize that Java had installed itself to this path)

[image]

I used both a netstat -a command and the Windows utility TCPView and I could not see any listening port opened by Java. What port number does this server normally listen on? Am I correct that it uses UDP?

david-schmidt, 2019-07-27:
What shows in the window when you push the "Ethernet" button?

Of the "surprising" install of Java, what bit-width is that one?  What about setting `ADTPRO_HOME` to a value you expect, ensuring it's 64-bit?

Do you have multiple Ethernet interfaces on your Windows machine?  It will have trouble deciding which to use if so.

amishrabbit, 2019-08-29:
Ran the program and then activated trace, and clicked Ethernet button:

8/29/19 8:53:11 AM Gui.getWorkingDirectory(): E:\Downloads\ADTPro-2.0.3\ADTPro-2.0.3\disks\
8/29/19 8:53:11 AM Gui.actionPerformed() exit.
8/29/19 8:53:13 AM Gui.actionPerformed() entry, responding to Ethernet; previous button is Disconnect
8/29/19 8:53:13 AM Gui.actionPerformed acting on the ethernet button.
8/29/19 8:53:13 AM Gui.startComms() entry.
8/29/19 8:53:13 AM Gui.startComms() didn't have an old thread to stop.
8/29/19 8:53:22 AM Gui.MenuAction.actionPerformed() responding to Quit
8/29/19 8:53:22 AM Gui.getWorkingDirectory(): E:\Downloads\ADTPro-2.0.3\ADTPro-2.0.3\disks\

david-schmidt, 2019-08-29:
Looks like nothing exciting happened.  Anything in the command prompt window?

amishrabbit, 2019-08-29:
Nothing happened at all, which has been the case from the beginning. It isn't clear that the app actually is fully running.

I did an experiment and installed the RXTX and that seems to work

david-schmidt, 2019-08-29:
This isn't something I've run across before.  rxtx can be problematic if you have a 32/64 bit mismatch between Java and OS, but Ethernet "just works."