ADTPro/adtpro #79: Win10 64bit. Apple IIc "No Serial Ports or RXTX Library is not installed"

Androdoxia, 2019-07-29:
Hi, I'm new to ADTPro and collecting older tech.

I've been trying to get this working for about two hours now but I'm pretty stumped.
I've ran the software, connected the cable to the port with the telephone symbol and then to a USB port on my laptop. Copying pretty much what I've seen people do on youtube.

I go to type the command line on the IIc (Except on this IIc there is no # just a £, however the button with all the boxes next to the 80/40 button changes it temporarily to a #). When I tell ADTPro on my Laptop to use a serial connection it just says "There are no serial ports, or the serial library RXTX is not installed or visible. Please see http://adtpro.com/install.html for details" except I'm not sure if I'm being dense, blind or stupid but I've not found any answers on that link.

david-schmidt, 2019-07-29:
I suspect the issue earlier was due to a missing driver on your host computer.

As for the X's - have you formatted your disks before trying to write them?  And, you're using double density (DD) and not high density (HD) disks, right?

Androdoxia, 2019-07-29:
Yeah I figured it would be something missing. But I didn't do anything different. I reinstalled java with the same one, 64bit latest version. Rebooted a couple of times etc. Moved to IIc cable from modem to printer then back to modem and it just... Worked? 

The X's occur when receiving data from the laptop. The disks are double density. 

The ADTPro software wrote successfully after a few attempts but if I try using it, it still says Error $24 when selecting serial on the menu.

It seems to be temperamental. Sometimes it receives with no errors. Then it does get errors etc. Other times it just complaints of a timeout. Or it just stops with no message given.

david-schmidt, 2019-07-29:
Hmmm, I don't really know how to address temperamentality.  What chipset does the USB adapter are you using have in it?

Androdoxia, 2019-07-29:
I don't know, there isn't any writing on the adaptors. I had to buy it on ebay from California because retrofloppy does not ship to the UK.

david-schmidt, 2019-07-29:
If you go to the Windows Device Manager, you should be able to find your COM port and it will tell you details about it.

Androdoxia, 2019-07-29:
USB Serial Port (COM3)
Manufacturer: FTDI
bits/s: 9600
Data bits: 8
Parity: None
Stop bits: 1
Flow control: None
Driver: FTDI, 16/08/2017 v2.12.28.0. Microsoft Windows Hardware Compatibility Publisher

david-schmidt, 2019-07-29:
FTDI is your chipset - and that's good news.  They don't get cloned like the Prolific chipset does, so it's not a likely source of problems.  Have you been successful in creating a disk at all yet that will boot?

Androdoxia, 2019-07-29:
Only the ADTPro disk. But it just flags up with an error when going to Serial.
I was going to make a diagnostics disk to test the computer itself and make sure it's working but I can't find one that's compatible with the IIc (I test all disks on a II emulator and none of the diagnostic ones I can find will work)

david-schmidt, 2019-07-29:
> But it just flags up with an error when going to Serial.

Can you instead boot the disk and quit to BASIC - and then do a `CAT` to see what's on the disk?

Androdoxia, 2019-07-29:
/ADTPRO.2.0.3
Name                          Type              Blocks            Modified
*PRODOS                    SYS                34             18-JAN-18
*BASIC                         SYS                21             30-AUG-16
STARTUP.SYSTEM        SYS                4               14-SEP-18
ADTPRO                      SYS                1                14-SEP-18
ADTPROAUD               SYS                1                14-SEP-18
ADTPROETH                SYS                1                14-SEP-18
ADTPRO.BIN                BIN                29              14-SEP-18
ADTPROAUD.BIN        BIN                35              14-SEP-18
ADTPROETH.BIN         BIN                39              14-SEP-18
Blocks Free: 100                                                  Blocks Used: 172

david-schmidt, 2019-07-29:
`$27` is a generic I/O error.  So it could be that your media is bad (or the drive is not always successful at writing).  What if you try either of the other ADTPro versions - audio or ethernet?  Do either of those start successfully?

Androdoxia, 2019-07-29:
No. They all give the same error. Which is weird cause the write had no errors

david-schmidt, 2019-07-29:
Pretty strange set of circumstances you have there.  If you have at least a ROM 0 IIc, you can run internal diagnostics:
https://apple2.fandom.com/wiki/Built-in_Diagnostics

Androdoxia, 2019-07-30:
Okay. I'm running the diagnostics now. How long does it normally take? It's been at it for 5-10min maybe. Just repeating a screen of lines/boxes with it changing from time to time.

david-schmidt, 2019-07-30:
It'll run forever; the fact that it's continuing means it didn't find anything wrong.

Androdoxia, 2019-07-30:
Ah okay then. I thought maybe it would stop and say it's fine. So where do I go from here?

david-schmidt, 2019-07-30:
Good question.  I don't have a concise enough problem statement, so I don't know what to suggest.  Maybe take a video of start-to-finish copying of a disk, and we can spot what's going wrong?

Androdoxia, 2019-07-30:
Okay then. I'll upload one to google drive and link it when it's done.
I tried writing to a disk with ADTPro but it again failed half way and said time out.

Androdoxia, 2019-07-30:
The floppies I use: https://drive.google.com/file/d/1--rxathj9dQ-vkluCy6JM9iTRMnvF8Kt/view?usp=drivesdk

The video: https://drive.google.com/file/d/1cbP6KS-09qx_OMwhshAosXU8gDRSEIkm/view?usp=drivesdk

david-schmidt, 2019-07-30:
That's a good walkthrough.  I can clearly see it successfully bootstrap and format disks, so I think everything on the Apple II side is fine.  Given that the serial port craps out halfway through, I'm left suspecting the USB serial port and/or its driver are at fault here.

Androdoxia, 2019-07-30:
Thanks. So would I be best buying a new adaptor? The driver has been installed and reinstalled. I don't know of any UK vendors for these adaptors. The only sure place I could of checked went bankrupt a couple of years ago.

david-schmidt, 2019-07-30:
Have you tried sourcing them on Amazon?

Androdoxia, 2019-07-30:
No to be honest. I tend to avoid Amazon since it can be pricey. There are a few USB to RS-232 cables but none of those adaptor heads. I'll try buying one.  Like say this one, https://www.amazon.co.uk/UGREEN-20210-Converter-Connecting-Controller-grey/dp/B00QUZY4UG/ref=sr_1_4?keywords=USB+to+RS-232&qid=1564507284&s=gateway&sr=8-4

david-schmidt, 2019-07-30:
Nope, PL Chipset=Prolific=Bad.  You get what you pay for.

Androdoxia, 2019-07-30:
Ah. Unfortunately almost all Amazon UK ones are using Prolific chipsets.

david-schmidt, 2019-07-30:
Keep looking... I found the right one in the "related" section of yours (maybe influenced by my own searches, though).

Androdoxia, 2019-07-30:
Ah. I did find one, towards page 12. I just kept searching for FTDI.
There were a few that claimed to be FTDI but in the reviews people mentioned they're actually prolific.
I'll order this FTDI one and get back to you with how it worked. I'll also try getting it to work on my PC encase its my laptop that is being weird. But the PC is being more of a pain to get working. FTDI Drivers are installed, Java Dev kit installed same as the laptop etc. But for whatever reason this PC is simply uninterested. Could be an age thing, this PC is very new compared to the 6 year old laptop.

david-schmidt, 2019-07-30:
A new PC will be fine.  Ensure you've got the 64-bit version of Java installed; Oracle automatic downloader will generally pick the _wrong_ one if left to its own devices.

Androdoxia, 2019-07-31:
I've got a cable on the way, coming between the 1st and 2nd.
However I've noticed a couple of things
1) ADTPro keeps "Stopping". I have to press disconnect then serial and then it seems to act like it remembers it was doing something. I've had to do this while doing a speedyboot.

2) The errors occuring while writing to disk are occuring both at write and receive. 

I suspect the issue could be the drive. It does make an unsavory sound similar to the sound it makes during boot when the error occurs.

david-schmidt, 2019-07-31:
Let's reserve judgement until the new cable arrives.  It sounds to me that it is something to do with the serial connection, and the most likely culprit so far is the USB cable/driver pair.

Androdoxia, 2019-08-04:
New cable arrived today. It is definitely a 50/50 issue.

New cable works flawlessly. No timeouts and 7 out of 10 disks wrote flawlessly.

The 3 disks that did fail, failed because they're bad disks. I/O Error etc. Seller claims Apple IIc can't read Double Density disks and he's "Genuinely surprised" I managed to get 7 done. Even though I can't find any documentation that says Apple IIc's can't read double density disks.

david-schmidt, 2019-08-05:
I think I saw that guy on Facebook - Dave Duke?  Yeah, you only want to use DS/DD disks.  HD will positively not work.  But of course you do need to format them before trying to write.

Good news on the cable.  That's what it is almost all of the time.

closed 2019-08-05 by david-schmidt