ADTPro/adtpro #80: David Schmidt - Working Sider to pass on.

Timelord0, 2019-08-04:
I got in contact with you (I think) once upon a time about the Sider hard drive.  You apparently had tried several, all of which were dead from catastrophic power supply failure.  

I have an Apple ][e system, tons of disks, two disk drives, amber monitor, and Sider hard drive.  Everything works.  The computer boots off the Sider and my father configured the thing for all manner of stuff, including two OSes (I think), games for us, and all manner of things.

He has passed away, and nobody is left to truly care for the keeping the system.

I am happy to pass this all on to you to hopefully further your work or at least give you something to play with.  The only caveat is how the devil to get this to you given it is fragile, heavy as heck, and would be stupid expensive to ship.  I am currently in Louisiana.  If you are interested, give shout.

Sider:
http://apple2.org.za/gswv/USA2WUG/FOUNDING.MEMBERS/HOME.PAGES/EDHEL/sider/

david-schmidt, 2019-08-05:
Yep - that was me!  I remember the Sider well.  I still have the SASI controller in my inventory, too.  I'd love to have a working specimen - that would be a first!

I can ensure the system lives on, for sure.  If you're willing to do the packing, I'm willing to pay the shipping.  What I'd suggest is pulling out the cards from the IIe and packing them separately (they just wave around inside the computer, yanking on their slots if you leave them plugged in).  

The monitor isn't worth shipping, but everything else is.  If you take the components to a local UPS shop, they can do the packing for you - and ensure things are not going to crash around against each other inside the box by using dense packing materials.  That'll be important with the Sider vs. the peripheral cards.  And again, I can pay for that.

david-schmidt, 2019-08-05:
Found our old discussion:
https://sourceforge.net/p/adtpro/discussion/supportarchive/thread/271d9ca0/?limit=25#6343

Timelord0, 2019-08-06:
Sourceforge!  I couldn't figure where the heck we spoke.  Anyways, yeah, I can get it packed up.  I'm in process of closing on house and moving, so this will likely happen in September.

Timelord0, 2021-07-05:
Life got away from me again, but I had a moment to get all the gear properly packed and ready to go out.  Weight is about 54lbs.  Box is 22x12x28.  Everything except the amber screen included.  Assuming you are still interested, I can send it out asap.  Unsure how to privately contact you for address/paypal/whatever.

david-schmidt, 2021-07-06:
That is splendid.  If you follow the link back to sourceforge, you can open my profile and hit the "send message" button.  That'll get directly to me.

Timelord0, 2021-07-11:
On its way.  I'm glad it is going to someone who can appreciate it. :)

Timelord0, 2022-03-01:
Shipment arrive ok? Any Sider progress made? :)

david-schmidt, 2022-03-02:
Absolutely, safe and sound - thanks very much for that.  It keeps me company near my feet while it waits patiently for the round tuit to appear.
[image]

Timelord0, 2022-03-02:
The system I sent with it was so by my dad with a nice menu. Sider on, let it get to speed and such, turn on Apple. Should boot right into menu, the sacrificing giving you some good hints at what is going on. I'm very glad it all arrived well, and I look forward to the magical day you have time. Waiting for that day myself on a few projects. 🤣

david-schmidt, 2022-03-02:
I _have_ had time to go through all of the floppies, and I found a handful that were unique enough to post to asimov.  So my thanks for those as well.