ADTPro/adtpro #81: ADTPro 2.0.3 exits after clicking serial icon

zxarr, 2019-08-07:
I managed to get the correct Java installed and ADTPro 2.0.3 starts, I can see the com port, then it just exits after I click 'Serial'.  I only have one COM port, COM6, using a USB cable.  ADTPro sees it OK, from what I can tell.

I'm attaching the trace and the error log to this.  Not sure what to do next.

[ADTProTrace.txt](https://github.com/ADTPro/adtpro/files/3474781/ADTProTrace.txt)
[hs_err_pid6752.log](https://github.com/ADTPro/adtpro/files/3474782/hs_err_pid6752.log)

The cable I'm using was sent to me from RetroFloppy.  It's a //c system where ADT 2.0.3 starts up without issue.  Running WIndows 10 with a Prolific USB-to-Serial adapter.  64-bit Java after removing and reinstalling the latest 64-bit version.

Thanks.

david-schmidt, 2019-08-07:
It's your USB cable.  Prolific is junk, sorry to say.

zxarr, 2019-08-07:
I was afraid of that.  Any recommendations on a good chipset for a USB to
serial cable?

Thanks!

david-schmidt, 2019-08-07:
FTDI.

zxarr, 2019-08-09:
[hs_err_pid14532.log](https://github.com/ADTPro/adtpro/files/3484333/hs_err_pid14532.log)

zxarr, 2019-08-09:
> FTDI.

So I have an FTDI adapter now and I'm still getting the same error.  It just closes.
[hs_err_pid14532.log](https://github.com/ADTPro/adtpro/files/3484340/hs_err_pid14532.log)

david-schmidt, 2019-08-09:
Can you try using (pointing to) a Java 8 VM instead of 12?

zxarr, 2019-08-09:
I did that and it didn't close this time!  It said 'Connected'. 

I'll attempt a disk transfer later this evening.

zxarr, 2019-08-10:
Success!  All working now.

Thanks!

closed 2019-08-10 by zxarr