ADTPro/adtpro #83: Issues with ADTPro bootstrapping and serial communication

Subrosians, 2019-08-17:
I'm actually having virtually the same issue as https://github.com/ADTPro/adtpro/issues/20 except I'm already using a cable purchased from http://retrofloppy.com/products/ as directed from the ADTPro site. I've tried 2 different computers, one running Windows 7 32bit and one running Windows 10 (1903) 64bit, both with the newest JAVA installed. I've tried 4 different USB to serial adapters (3 different IOGear GUC232A adapters and a different Sabrent one with no model number on it). I have also confirmed that the configuration of the IIgs matches the defaults specified in the ADTPro instructions.

If I follow the instructions exactly, there is no output on the IIgs at all but if I enable "Apple IIc w/ Imagewriter Cable", I get a lot of garbage on the screen. It also seems that if I leave the serial connection closed on the Windows side, I get garbage on the screen as well. I have also tried getting the serial cable as far away from power as possible, in case there was some interference from that. 

Any ideas? At this point, I'm either thinking defective cable or bad serial port on IIgs.

david-schmidt, 2019-08-17:
It matters what you mean exactly by “latest” with java.  It’s been shown on other issues that Java 12 does not work; be sure you’re using Java 8.

Subrosians, 2019-08-17:
Sorry, should have been more descriptive, I used Version 8 Update 221.

david-schmidt, 2019-08-17:
If you’ve used multiple host machines and USB adapters, the GS is the most likely culprit.  The only place actual data would come from when things are otherwise quiet is the USB adapter or the GS.  The 8-pin null modem is passive.

Subrosians, 2019-08-18:
Although the IOGEAR GUC232A adapters have been reliable for me in the past (mostly with network switches and video encoders), they are not actually FTDI as the site recommends. Just to be 100% sure, I've ordered 2 different FTDI USB to Serial (RS232) adapters which should be in on Monday. I will update you then with my testing results.

Subrosians, 2019-08-20:
So, I tried with two different FTDI adapters (Sabrent CB-FTDI and the Ugreen 20206) and both are giving me the same results: Garbage when the serial port is closed, nothing when the serial port is open unless I enable "Apple IIc w/ Imagewriter Cable" which then gets me garbage while it is sending data. 

Reading online, I found someone that tested using the printer port and IN#1 to get to the first part of the bootstrap (which i know it can't do the virtual slot #8 thing), and when I tested it out, it worked perfectly (of course I can't use this because I can't finish the bootstrap). 

Now, I'm 100% sure it can't be the computers or serial adapters, and I think we can rule out the cable due to it working on the printer port and because you say that the cable can't introduce garbage, and that leaves me with the IIgs being the problem. I went into the IIgs configuration and again confirmed everything is configured as default. Anything else you could recommend me try at this point? Is this just a lost cause?

david-schmidt, 2019-08-20:
It does sound like port 2 on your GS is bad.  I've not had good luck using port 1 on the GS, so it might be time to look at finding another machine.

Subrosians, 2019-08-21:
So, wonderful news! I was able to borrow a Super Serial Card II from an Apple IIe that is currently unused and put it in slot 2 of the IIgs. After configuring Slot #2 to "Your Card", I was able to bootstrap perfectly and write a disk image with no problems! Now, I just need to buy another Super Serial Card II to put back in the IIe. Thanks for all of your quick replies!

david-schmidt, 2019-08-21:
Awesome, so it is the GS port 2.  That's kind of unusual, but I also started out with a bum port.  I bought a pile of spares later, so that's no longer an issue. ;-)  Congrats on getting that working!

closed 2019-08-21 by david-schmidt