ADTPro/adtpro #86: Unable to receive

MaxMan19, 2019-09-08:
I am trying to use ADTPro over an audio connection to transfer disk images from a modern computer (MacBook Pro) to 5.25" floppies on an Apple II+. The two computers seem to be able to communicate with one another, but when I try to receive the image on the II+, it stops after block 16 and returns to the ADTPro menu. There is no error message or any other indication of what went wrong. This has happened with several disk images. Am I doing something incorrectly?

david-schmidt, 2019-09-09:
It doesn't sound like it - but it is strange that it quits at the exact same spot.  Can you try adjusting the volume level, and ensure you aren't getting any stray sounds (like arriving mail, etc.) from the operating system?  Audio is very finicky and is susceptible to any ambient noise in the system.

MaxMan19, 2019-09-10:
Thanks for your reply. I have indeed already tried adjusting the volume level, and have verified that no other sounds are present.