ADTPro/adtpro #87: Boot strapping never completes on IIgs

amb1979i, 2019-10-17:
I am experiencing a problem with Prodos bootstrapping on my IIgs.  I have gotten this to work over the past two weeks but now I am unable.

grub2.dmp transfers without problem.  The ADTPRO Serial Bootstrapper splash screen appears.  However, it states "Found Slot 8" in stead of slot 2.  It remains on this screen and the file PDSpeed.raw continuously transfers to the IIgs.

Bootstrapping never completes.

I am using the modem port of the IIgs with the control panel for slot 2 set to modem.

Any ideas or troubleshooting advice?

amb1979i, 2019-10-17:
Update: I was successful when sending PRODOS first then the ADT program.

david-schmidt, 2019-10-17:
It should work either way.  But "Slot 8" is normal for a GS - it just means the built-in serial port as opposed to a physical SSC card plugged in to slot 2.

closed 2019-10-17 by david-schmidt