ADTPro/adtpro #88: ADTPro issue

thanospl68, 2019-10-20:
Hello everybody in the forum

I have an Apple IIe  European (PAL) and I bought a SSC Card just to run ADTPro to load few images to floppies

On the other end I 'm using an Intel Pentium PC with Windows 10 64bits and one serial port, all Java software manualy installed (64bits) and manual file mooving required has been done (Serial Drivers)

Server ADTpro runs ok and show the COM1 port on the Configuration.

Serial cable is a straight through cable DB9F to DB25M

I have a floppy came with the Apple IIe with the ADTPro client and seems to run ok.

When I try to receive an image to Apple using ADTPro client (D)ir  or (R)eceive (or any other) command ends with a message like waiting for response, ESC to cancel that last forever

When I hit ESC Apple just beeps.

So far everyone can suggest that the SSC card and/or the serial cable is at fault but the connection works fie when I send Bootstrap (Speedibot etc) from the PC with all Apple II commands required (IN#1, control A, SSC prompt, 14B etc) and seems that it communicate and transfer fine.

I did try a different cable  but it didn't work either, this time not even Bootstraping

 

Do you have any suggestions? I was thinking to buy and use a USB to Serial Adapter but there are compatibility problems there also

 

Thanks in advance

thanospl68, 2019-10-20:
I also need to say that bootsrapping does stops at Loading MLI without have the flashing character loading thing

david-schmidt, 2019-10-20:
The circumstances sound like your Apple II's SSC can receive, but not send.  That would account for Speediboot not working beyond the initial load (which only requires receive to function).  It waits for MLI because at that point, it has sent a command from the Apple II to the PC, and is waiting for a response.  But since send isn't working, the PC never "hears" it.  It is not unusual for one side of an SSC to work while the other is broken, so as long as you're 100% sure how your 9-25 cable is wired, I'm inclined to suspect the SSC.

thanospl68, 2019-10-20:
David thanks for your fast reply, I can make my own cable so do you know the exact connections needed for ADTPro to work  with a DB9F to DB25M ? 
Thanks again
Tom

david-schmidt, 2019-10-21:
Here's the standard 9-25 pin mapping for straight-through:
```
Pinout:
1-8
2-3
3-2
4-20
5-7
6-6
7-4
8-5
9-22
G-G
```

thanospl68, 2019-10-24:
Thanks David, my cable is exactly like that, but it didn't work 
I did make a loopback  DB25M connector to test the SSC and it seems to work both ways since it repeats the characters i type back to the card and the screen (Echo Enable), 
finally I decide to make a plain RX -TX-Ground  (No Handshake signals) cable to connect my PC to Apple
DB9     DB25
Rx 2 -> Tx 2 
Tx 3 -> Rx 3
GND 5-> GND 7
And it works fine with bootstraping and ADTPro even at 115kbbs
Seems that Handshake signals have timing or other problems at least in my case
Thanks 
Tom

david-schmidt, 2019-10-24:
Great - glad you got it working!

closed 2019-10-24 by david-schmidt