ADTPro/adtpro #90: Checksum/Disk Error

pxlbluejay, 2020-01-11:
Whenever I try to load ANY disk file, I get either a checksum or disk error. It goes all the way up to the Sector/Track graph thing, but at the step of Inflating/writing the file, it shows the checksum or disk error. I am using  the correct kind of disks(DSDD Apple formatted). I really need to get a new way to get my Apple //e running DOS since my real copy of PRODOS is barely working. Help!

david-schmidt, 2020-01-12:
I can't quite tell what you're doing based on the words you're using.  I get the feeling you might be using ADT vs. ADTPro.  Can you maybe take a video of your apparatus, and show the the steps you think should get you to writing a disk?

pxlbluejay, 2020-01-16:
Uhh... Actually, nevermind. I think I've solved the problem. I think it was
me making a stupid mistake:)

closed 2020-01-22 by david-schmidt