ADTPro/adtpro #91: Blank screen after loading ADTPRO (serial)

StuartSeeley, 2020-01-13:
I'm seeing a blank screen after loading ADTPRO that I believe is not related to cabling or the windows software.  I have two seemingly identical Apple IIgs units here. One of them is able to load ADTPRO fine and make disks, the other gives me a blank screen and seems to hang.
If I hit ESC it makes some beeping.. I *CAN* get to control panel/reboot/etc. so I'm not sure what to think here.  Control panel options for everything including serial ports are set the same between systems.

david-schmidt, 2020-01-13:
So, using identical cabling between the two machines, you see one hang during ADTPro startup?  Sounds like you've positively identified a hardware fault in that machine.

StuartSeeley, 2020-01-14:
Yeah, that is what I feared... the computer in question seem to work fine otherwise though so I wasn't sure... was hoping you could suggest what might cause it anyway. It literally is the exact same wires, only the IIgs itself is being swapped out. One works, one has a blank screen when adtpro loads. Both can boot AppleII games.

StuartSeeley, 2020-01-14:
Got it to work. I suspect it was due to Appletalk being on instead of printer port... but not 100% sure.
Thanks though. All the best.

david-schmidt, 2020-01-14:
Hmmm, that doesn't make a lot of sense - you might have mistakenly used the printer port vs. the modem port - discussed here:
https://adtpro.com/connectionsserial.html#MiniDIN8
But even if you had Appletalk enabled in the control panel (which uses the printer port) it wouldn't have affected your use of the modem port.  Glad you got it going!

closed 2020-01-14 by david-schmidt

StuartSeeley, 2020-01-17:
Not sure if I can reply to this to get you a message....

But I've confirmed that is the behaviour on both of my GS's here.
If Apple talk is turned on and printer port is turned off in the control
panel, adtpro will hang with a blank screen.
I was not using the modem port.

Anyway, yes fully working now that I figured that out!
Thank you for the wonderful stuff! Cheers!

Stuart

david-schmidt, 2020-01-17:
Ah, very interesting.  Now that I think of it, Appletalk is probably going to work over the SCC chip pretty hard, competing for attention with ADTPro.  I guess I'll have to punt this back to the connections page, which does show what the defaults should be for the control panel. ;-)

StuartSeeley, 2020-01-17:
Yeah, I couldn't find any one who had a solution anyway... so would be a
good tip to put in your documentation.
Anyway, thanks again, all the best!

Stuart