ADTPro/adtpro #92: Cannot get serial connection to "bite"

tfrazier-home, 2020-02-06:
Hey there,
I am so excited to get back into the Apple II world again!!!. I started my programming career back in 1985 after buying an Apple IIC in 1985. Now after 35 years in software development, as McArthur saids, "I have returned... to buy another Apple IIc off of eBay!!!

I have got your disk, also read and followed the online docs videos and I just can't get the darn connection to "bite" between my MacBook Pro and the IIc.

I am using a connection that the seller claimed came from the Retrofloppy site but I am not sure it is good. The port seems to respond to the IN#2 and ctrl-B okay and the ADTPro side displays the connect ungreyed but when I send it across nothing appears on the IIc screen.

And I am at a loss as to confirm the macbook is actually sending it. I wonder if I could sniff the USb port? 
Any help much appreciated!!!

Tim

david-schmidt, 2020-02-06:
Hi, Tim -

Welcome back!  This is exciting for sure.  So, RetroFloppy cables are the Gold Standard, so that shouldn't be the problem.  But... on the USB end, have you installed the appropriate driver on the Mac side?  What is the list of devices available on the Mac/Server side of ADTPro for serial ports when you've got your USB adapter plugged in vs. not plugged in?

tfrazier-home, 2020-02-07:
Hi David,
Thanks!! I believe I downloaded what was asked for - the Java SDK I believe.
When I check the serial connection through ADTpro I see two ports I think are usb but they are incoming. Shouldn;t I have an outgoing?
See atta
[image]
ch

david-schmidt, 2020-02-07:
You not only need the Java SDK (which looks like it's working fine), but also the driver for your USB serial adapter.  You'd see it listed in the box along with your Bluetooth Incoming ports if it were installed correctly.  Can you send a snap of the USB-RS232 adapter you've got?

tfrazier-home, 2020-02-07:
hmm, yea I got the cable but no driver from the seller. Didn't see any driver downloads in the docs (but I can't see a lot of things anymore)
Here's a pic of the cable.
[image]

tfrazier-home, 2020-02-07:
Find it!!! I think! Installing now

tfrazier-home, 2020-02-07:
oops FOUND IT!

schmidtd, 2020-02-07:
Ok.  The beige cable is from RetroFloppy; the BENFEI USB-RS232 one is not.  I'm not sure what chipset it might be using, so that will be of concern when finding the driver for Mac.

tfrazier-home, 2020-02-07:
I found it . You have to use the Prolific driver!!!
Google search:
prolific usb to serial driver mac

http://www.prolific.com.tw/US/ShowProduct.aspx?p_id=229&pcid=41

download zip file at the top

tfrazier-home, 2020-02-07:
The screen filled with chars and ADTPro downloaded and booted. Also was able to access a file and attempt an copy to drive 1. 
Now all I have to do is get the drive working!!!
"I can't get anything to format or copy a disk  successfully. 
Even Locksmith v6.0 can't copy a disk.
Even after using Locksmith to adjust speed.

I think it needs a "reallignment" or the write head is totally screwed.

tfrazier-home, 2020-02-07:
Thanks so much for your help!!! I love it when something goes right!!!
:-)

david-schmidt, 2020-02-07:
Check the kind of blank disk you're using.  They need to be double density, and not high density.  And of course you'll need to be able to successfully format them before you can write to them.

tfrazier-home, 2020-02-08:
Yep I know. I bought some on amazon - and now I don't trust whether they are REALY DS,DD or not. I will order from retrofloppy

david-schmidt, 2020-05-03:
Closing - surely all is well now!

closed 2020-05-03 by david-schmidt